olikraus/u8g2 #2309: u8g2 noname API mean?

WangShuoran, 2023-11-30:
I am a roob, I want to know the u8g2 use method, but I have searched as much information as possible, but there is no direct answer.

Thank you for answer these simple questions, but this is very difficult for me.

What is the U8g2 in ssd1306 `U8G2_SSD1306_128X64_NONAME_F_HW_I2C` noname mean?

And I want to know, When I search CH1116 and SH1106 in issue, I find any question, but SH1106 in csrc directory do not have 128*64, and CH1116 never exist. But CH1116 and SH1106 have almost same commands  with SSD1306.

So I want to know does the SSD1306 in u8g2's use don't have scrolling command?

And the charge pump command is not same, How can I find init API in u8g2 and modify it?

olikraus, 2023-11-30:
> What is the U8g2 in ssd1306 U8G2_SSD1306_128X64_NONAME_F_HW_I2C noname mean?

This is a constructor, needed to init your display

>  but SH1106 in csrc directory do not have 128*64

SH1106 is in the same files like the SSD1306

>  CH1116 never exist

see https://github.com/olikraus/u8g2/issues/2030

> So I want to know does the SSD1306 in u8g2's use don't have scrolling command?

U8g2 does not support scrolling, but you can send custom command to do this, see sendF command: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf

WangShuoran, 2023-12-01:
Thank you! And does Is it unnecessary to manually change pages after completing a page in SSD1306? But CH1116 and SH1106 require manual page changing.

Does u8g2 completing a page with change pages?

WangShuoran, 2023-12-01:
What is the U8g2 in ssd1306 U8G2_SSD1306_128X64_NONAME_F_HW_I2C noname mean?

This is a constructor, needed to init your display

I want to know why you named noname, and other driver file not have this keyword

closed 2023-12-01 by WangShuoran