olikraus/u8g2 #2310: Blurry/tall text using AliExpress 240x128 3.37" SSD1322 OLED

olikraus, 2023-12-02:
### Discussed in https://github.com/olikraus/u8g2/discussions/2308

<div type='discussions-op-text'>

<sup>Originally posted by **TomsJensen** November 29, 2023</sup>
Hey,

I set up a few of the U8G2 examples using an 3.37" OLED using both the the `U8G2_SSD1322_240X128_F_4W_SW_SPI` and `U8X8_SSD1322_240X128_4W_SW_SPI` constructors, and to my surprise they fired up and _did_ actually render to the screen. The only thing is that the text reads a bit blurry. It appears if lines being drawn horizontally are being smeared across two rows. I'm wondering if there is some small setting or fix that I'm forgetting that would resolve this.

Thanks!

[image]
[image]</div>

olikraus, 2023-12-02:
ToDo: Create new constructor and takeover the setup values from the discussion.

olikraus, 2023-12-02:
@https://github.com/TomsJensen

I have created a new beta release 2.35.8. The constructor name starts with U8G2_SSD1322_TOPWIN_240X128
Can you check whether this works?

Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2024-08-24:
hmmm no feedback, closing

closed 2024-08-24 by olikraus

TomsJensen, 2025-03-19:
Hello! Somehow, I wasn't notified of further activity over here, so I apologize. I had a moment to test the code with the original display today, and it appears to be working just fine. Thanks for including the new header. <3