olikraus/u8g2 #2311: u8g2_font_wqy16_t_gb2312 problem in Arduino Mega2560 and LCD12864

VanishingDice, 2023-12-04:
(please forgive me for my grammar mistake...)
I'm using Arduino Mega 2560, I want to print chinese characters on LCD12864, when I use u8g2_font_wqy16_t_chinese1, 2 and 3, everything goes well, but characters isn't enough. when I trying to use u8g2_font_wqy16_t_gb2312 or 2312a, b, when the complier finished all and trying to combine everything together, the problem will happen like this:

```
将所有内容链接在一起。。。
C:\Users\Admin\AppData\Local\Temp\ccuadcHM.ltrans0.ltrans.o: In function `u8g2_font_get_byte':
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:114: undefined reference to `u8g2_font_wqy13_t_gb2312b'
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:114: undefined reference to `u8g2_font_wqy13_t_gb2312b'
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:114: undefined reference to `u8g2_font_wqy13_t_gb2312b'
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:114: undefined reference to `u8g2_font_wqy13_t_gb2312b'
C:\Users\Admin\AppData\Local\Temp\ccuadcHM.ltrans0.ltrans.o: In function `u8g2_SetFont':
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:1287: undefined reference to `u8g2_font_wqy13_t_gb2312b'
C:\Users\Admin\AppData\Local\Temp\ccuadcHM.ltrans0.ltrans.o:c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib/u8g2_font.c:1287: more undefined references to `u8g2_font_wqy13_t_gb2312b' follow
collect2.exe: error: ld returned 1 exit status

使用 2.34.22 版本的 U8g2 库，在列出的文件夹中：C:\Users\Admin\Documents\Arduino\libraries\U8g2
使用 1.0 版本的 SPI 库，在列出的文件夹中：C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI
使用 1.0 版本的 Wire 库，在列出的文件夹中：C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire
exit status 1

Compilation error: exit status 1
```

I noticed that the charset u8g2_font_unifont_t_chinese1 and 2 can work, but chinese3 in unifont doesn't work, that's similar to #510 , but that is a really old issue, and seems that person used a customized charset to solve the problem, is there any possibility to just use standard gb2312 charset on arduino mega2560?

this is my code:

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, 4, 3, 2, 5);

void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();
}

void loop(void) {
  u8g2.setFont(u8g2_font_wqy13_t_gb2312b); 
  // u8g2.setFont(u8g2_font_unifont);
  u8g2.setFontDirection(0);
  u8g2.clearBuffer();
  u8g2.setCursor(0, 15);
  u8g2.print("Hello World!");
  u8g2.setCursor(0, 40);
  u8g2.print("你好世界");		// Chinese "Hello World" 
  u8g2.sendBuffer();
  
  delay(1000);
}
```

VanishingDice, 2023-12-05:
Thank you! What a silly mistake I made! As you said, the array is too big for Arduino:
```
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:111831:15: error: size of array 'u8g2_font_wqy12_t_gb2312' is too large
 const uint8_t u8g2_font_wqy12_t_gb2312[202690] U8G2_FONT_SECTION("u8g2_font_wqy12_t_gb2312") =
               ^~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:118175:15: error: size of array 'u8g2_font_wqy12_t_gb2312a' is too large
 const uint8_t u8g2_font_wqy12_t_gb2312a[109189] U8G2_FONT_SECTION("u8g2_font_wqy12_t_gb2312a") =
               ^~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:121597:15: error: size of array 'u8g2_font_wqy12_t_gb2312b' is too large
 const uint8_t u8g2_font_wqy12_t_gb2312b[118722] U8G2_FONT_SECTION("u8g2_font_wqy12_t_gb2312b") =
               ^~~~~~~~~~~~~~~~~~~~~~~~~
```

the array limit for Arduino is 32768 elements, gb2312 and unifont_chinese3 are both out of this limit, so they can't work on Arduino. Is there any possibility to put them in the program space to use? or I should create a customized charset?

olikraus, 2023-12-05:
> so they can't work on Arduino.

It is not related to "Arduino". It is the AVR architection which adds this limitation. I mean the problem is the "Arduino Mega2560" controller. If you would use ESP32 or an ARM based Arduino Board than this limitation doesn't exist.

> Is there any possibility to put them in the program space to use? 

All fonts are already located in PROGMEM space (this is enforced by the U8G2_FONT_SECTION macro).

> I should create a customized charset?

Of course this is an option. There is a video on this topic:
https://github.com/olikraus/u8g2/blob/34be7c84883a3ffc927e04eda7a99c539da040ee/doc/faq.txt#L284-L286

VanishingDice, 2023-12-08:
Thank you very much! This helps me a lot!

closed 2023-12-08 by VanishingDice