olikraus/u8g2 #2316: use signed int 8bit with MUI ???

reza6707-reza6707, 2023-12-08:
Hi everyone.
 
I wrote a program for working with Ks0108 and a rotary encoder for change menu .

In my program i need get a signed number with this range -10 and +10

i wrote 
// Return Value of Temperature Sensor Calibration 
int8_t sensor_calibration = 0 ;

I used this var in MUIF_U8G2_MIN_MAX()

MUIF_U8G2_U8_MIN_MAX("SC", &sensor_calibration, -10, 10, mui_u8g2_u8_min_max_wm_mud_pi),

so when compile code in arduino ide and start simulation in proteus and test code ....... result is this 

https://github.com/olikraus/u8g2/assets/61326080/602caa0c-673e-4513-897c-a2ffc8115dc3

olikraus, 2023-12-09:
maybe I can add signed int once in a while...

reza6707-reza6707, 2023-12-15:
tnx for replay .

Is it possible to upgrade the "MUIF_U8G2_U8_MIN_MAX" function to receive float numbers??

olikraus, 2023-12-15:
Hmm... Once signed ints are supported, I suggest to provide multiple fields to the use.
I mean, the user could enter the float like this:  x*10^y
Both x and y would be signed int. Then on the form there would be two fields: one for x and one for y.
From that information you could build the float number

olikraus, 2024-08-27:
``` C++
uint8_t mui_u8g2_s8_min_max_wm_mse_pi(mui_t *ui, uint8_t msg);
uint8_t mui_u8g2_s8_min_max_wm_mud_pi(mui_t *ui, uint8_t msg);
uint8_t mui_u8g2_s8_min_max_wm_mse_pf(mui_t *ui, uint8_t msg);
uint8_t mui_u8g2_s8_min_max_wm_mud_pf(mui_t *ui, uint8_t msg);
```

closed 2024-08-27 by olikraus