olikraus/u8g2 #2318: Can be adapted to SSD1363 256*128

pipi-1, 2023-12-09:
It's pretty much the same as a 13xx controller
void OLED_Init(void)
{

	OLED_RES_Clr();	
	delay_ms(10);	
	OLED_RES_Set();

	OLED_WR_REG(0xfd);  /*Command Lock*/ 
	OLED_WR_Byte(0x12);

	OLED_WR_REG(0xAE);
[SSD1363.pdf](https://github.com/olikraus/u8g2/files/13622824/SSD1363.pdf)

	OLED_WR_REG(0xC1);  
	OLED_WR_Byte(0xFF);	

	if(USE_HORIZONTAL)
	{
		OLED_WR_REG(0xA0);  
		OLED_WR_Byte(0x24);	
		OLED_WR_Byte(0x00);	

		OLED_WR_REG(0xa2);////A2.Display Offset
		OLED_WR_Byte(0x80); ////Offset:: 0~127				
	}
	else
	{
		OLED_WR_REG(0xA0);  
		OLED_WR_Byte(0x32);	
		OLED_WR_Byte(0x00);	

		OLED_WR_REG(0xa2);////A2.Display Offset
		OLED_WR_Byte(0x20); ////Offset:: 0~127		
	}
	

	OLED_WR_REG(0xca);////CA.Set Mux Ratio
	OLED_WR_Byte(0x7f); ////Mux:: 0~255

	OLED_WR_REG(0xad); ////AD.Set IREF
	OLED_WR_Byte(0x90); ////Select:: Internal

	OLED_WR_REG(0xb3); ////B3.Set Display Clock Divide Ratio/Oscillator Frequency
	OLED_WR_Byte(0x61); ////DivClk:: 0~255

	OLED_WR_REG(0xb9); ////B9 Default GAMMA

	OLED_WR_REG(0xAF); //Set Display On

}
This is the official initialization
void Column_Address(u8 a,u8 b)
{
	OLED_WR_REG(0x15);       // Set Column Address
	OLED_WR_Byte(a+0x08);
	OLED_WR_Byte(b+0x08);
}

//******************************************************************************
//    函数说明：OLED显示行的起始终止地址
//    入口数据：a  行的起始地址
//              b  行的终止地址
//    返回值：  无
//******************************************************************************
void Row_Address(u8 a,u8 b)
{
	OLED_WR_REG(0x75);       // Row Column Address
	OLED_WR_Byte(a);
	OLED_WR_Byte(b);
	OLED_WR_REG(0x5C);    //写RAM命令
}

pipi-1, 2023-12-09:
My own changes to the initialization and offset will have some ghosted characters
[image]

pipi-1, 2023-12-09:
I looked at #2298, modified 8to32 function successfully solved
[image]

olikraus, 2023-12-09:
The links do not work and I also do not understand how I can help here.

pipi-1, 2023-12-09:
> The links do not work and I also do not understand how I can help here.链接不起作用，我也不知道我怎么能在这里帮助。
This issue has been resolved

pipi-1, 2023-12-09:
But you can improve it. ssd1363 256*12，Hope you can help
[1363.zip](https://github.com/olikraus/u8g2/files/13623509/1363.zip)

olikraus, 2023-12-10:
> But you can improve it. ssd1363 256*12，Hope you can help

I can only help or improve if you tell me the problem.