olikraus/u8g2 #2319: u8x8log top line glitching / turns to garbage characters

JamesHamilton4, 2023-12-09:
I love u8x8log but I was having a problem where the top line of characters would turn to garbage as it scrolled vertically.
As I was transmitting via radio I assumed it was something to do with corrupted data but it turns out u8x8log (with my setup at least) does not like blank lines. Am using a 128x64 OLED with  I2C.
So if you want a blank line in your scrolling text do not use either:
u8x8log.print("\n");
Or
u8x8log.println();
but instead use u8x8log.println(" ");
Not sure if this is a bug or something to do with my setup but anyway I hope that saves someone some time and thanks so much to the creator of u8x8log.

olikraus, 2023-12-09:
hmm can you provide a more complete example which reproduces this issue?

JamesHamilton4, 2023-12-09:
`
#include <Arduino.h>
#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C

#define U8LOG_WIDTH 16
#define U8LOG_HEIGHT 8
uint8_t u8log_buffer[U8LOG_WIDTH*U8LOG_HEIGHT];
U8X8LOG u8x8log;

void setup(void)
{
  
  u8x8.begin();
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  
  u8x8log.begin(u8x8, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer);
  u8x8log.setRedrawMode(1);   // 0: Update screen with newline, 1: Update screen for every char  
  
}

unsigned long t = 0;

void loop(void) {
  if ( t < millis() ) {
    t = millis() + 15000;     
    u8x8log.print("\f");     
  }
  u8x8log.print("millis=");
  u8x8log.print(millis());
  u8x8log.print("\n");
  u8x8log.print("\n");
  delay(500);
}`

JamesHamilton4, 2023-12-09:
I've recreated the problem with your simple example, because my own code was too long.
By removing the second instance of  "u8x8log.print("\n");" everything is ok.
But to have two "u8x8log.print("\n");", to create a blank line, the glitches appear

JamesHamilton4, 2023-12-09:
Same problem occurs if you use to u8x8log.println(); to create the blank line.
For me, I can only use u8x8log.println(" "); to create a blank line without glitches.

JamesHamilton4, 2023-12-09:
Also, in case it's information that is of use, if I change:
#define U8LOG_HEIGHT 8
to
#define U8LOG_HEIGHT 7
the glitches appear at the bottom, rather than the top. But this screen with this font can fit 8 lines.

Herwig9820, 2023-12-10:
Hello, just like Hamilton4 I also love u8x8log! And I also have the same issue when printing to an OLED (128 x 64 pixels). 
Printing a CRLF sequence without anything else makes the oled screen behaving strangely.
A small program demonstrating this:

`
#include <Arduino.h>
#include <U8g2lib.h>        // https://github.com/olikraus/u8g2  
#include <Wire.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8;
U8X8LOG u8x8log;        

// Define the dimension of the U8x8log window
#define U8LOG_WIDTH 16
#define U8LOG_HEIGHT 8

// Allocate static memory for the U8x8log window
uint8_t u8log_buffer[U8LOG_WIDTH * U8LOG_HEIGHT];

void setup(void)
{
    // Startup U8x8
    u8x8.begin();
    u8x8.setFont(u8x8_font_chroma48medium8_r);
    u8x8log.begin(u8x8, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer);
    u8x8log.setRedrawMode(0);   // 0: Update screen after a newline  

    int i{ 0 };
    while (true) {
        u8x8log.print(++i);
        u8x8log.print(">> ");
        u8x8log.print(millis());
        u8x8log.print("\r\n");
        u8x8log.print("\r\n");

       delay(1000); 
    }
}

void loop(void) {
    // nothing here
}

`

Best regards

olikraus, 2023-12-12:
I hope i find time too look into this soon.

olikraus, 2024-08-31:
Thanks for reporting this. It will be fixed with the next release.

closed 2024-08-31 by olikraus