olikraus/u8g2 #2320: Difference between SSD1306_NONAME and SSD1306_1F? (Wiki update)

olikraus, 2023-12-10:
### Discussed in https://github.com/olikraus/u8g2/discussions/2317

<div type='discussions-op-text'>

<sup>Originally posted by **egnor** December  8, 2023</sup>
I am using a [64x32 OLED from buydisplay/EastRising](https://www.buydisplay.com/white-0-49-inch-oled-display-64x32-iic-i2c-ssd1306-connector-fpc) using I2C and had a bunch of struggle before realizing that `SSD1306_NONAME` doesn't work for me (garbled junk on screen) and `SSD1306_1F` works perfectly.

In code, the main difference is that the `_1F` flavor uses a multiplexing ratio value of 0x1f (hence the name?), `_NONAME` uses 0x2f. That corresponds to 32-way row driving (0x1f) or 48-way row driving (0x2f). I can't imagine why a 64x32 OLED would use 48-way row driving, but... the world is full of strange things...

I'm now gathering that accepted practice is just to try all variants of a particular display to see what works. Perhaps this could be noted in the "Setup" documents?

If the situation is clarified, I'm happy to edit the wiki / code as needed to save others similar confusion.</div>

olikraus, 2023-12-10:
see discussion for details

egnor, 2023-12-11:
Proposal copied over from the discussion: Edit the `u8g2setupcpp` wiki page as follows, with my additions marked by 🆕 (delete the mark before merging!):

---

## Constructor Name

The name of the Arduino C++ constructor has the following parts:

| No | Description | Example |
|---|---|---|
| 1 | Prefix | `U8G2` |
| 2 | Display Controller | `UC1701` |
| 3 | Display Name | `DOGS102` |
| 🆕 4 | 🆕 Variant (optional) | 🆕 `ALT` |
| 5 | Buffer Size | `1`, `2` or `F` (full frame buffer) |
| 6 | Communication | `4W_SW_SPI` |

These parts are connect with `_`. The full constructor name for the example will be `U8G2_UC1701_DOGS102_ALT_1_4W_SW_SPI`. All available constructor names are listed in the [reference](#reference) section below.

🆕 If multiple variants are available for a display name, you may need to experiment to find the one that works for you. Small unbranded displays often come in "flavors" that have important differences but aren't marked on the packaging!

---

Those edits should apply to `u8x8setupcpp` as well. Similarly, I'd edit the `u8g2setupc` page like this:

---

## Setup Naming Convention

Each setup procedure name has the following naming convention:

`u8g2_Setup_<display>_🆕<variant>_<i2c>_<memory>`

 * `<display>`: Describes the display name and subtype (dimensions, branding).  
 * 🆕 `<variant`>: (Optional) Describes a different way to configure the driver.
 * `<i2c>`: I2C protocol requires special care. If the display supports I2C communication interface, 
  then there is a special setup procedure for I2C, which includes the string "i2c" in the procedure name.
  For SPI or parallel communication, use the setup procedure without "i2c" in the procedure name.
 * `<memory>`: This is `1`, `2` or `f` for one page, two page or full page mode.

🆕 When multiple variants (`_ALT`, `_1F`, etc.) are available, you may need to experiment to find the one that works for you. Small unbranded displays often come in "flavors" that have important differences but aren't marked on the packaging!

In any case, the naming convention is just a rough guide; the actual function names and their possible arguments are listed below.

---

olikraus, 2024-04-14:
That _1, _2, _F should have been mentioned, but I like the addon from @egnor

olikraus, 2024-08-24:
done...

closed 2024-08-24 by olikraus