olikraus/u8g2 #2323: Is the library supported on arduino nano rp2040 connect?

Mordechay0000, 2023-12-13:
I tried to operate my lcd with 'Arduino nano rp2040 connect' and it doesn't work, with 'Arduino uno' it works. Could it be that the library is not supported by rp2040 connect or is there another reason?

olikraus, 2023-12-13:
Did you use the https://github.com/earlephilhower/arduino-pico extension?

Mordechay0000, 2023-12-13:
it's not good?:

[image]

olikraus, 2023-12-14:
Well... it should be ok. It is like this: U8g2 will only use the standard calls to the Arduino Framework. From my perspective those calls work in the same way for many different controller types. So I have tested this vor AVR (like Arduino Uno), ARM, Espressor and STM32 architectures. So I2C (wire), SPI and GPIO interfaces are all the same for those architectures and u8g2 does not require any changes for this. It "just" works. In other words, my assumption is, that the above mentioned arduino-pico framework is not (yet) compatbile with all the other frameworks. 

The other option might be, that you still have a wire issue with the board...

egnor, 2024-04-13:
For reference, there are two Arduino cores for the rp2040 chip (the brains of the Pico): [the official Arduino Mbed-based core](https://github.com/arduino/ArduinoCore-mbed), and [Earle Philhower's community core](https://github.com/earlephilhower/arduino-pico).

I can confirm that Earle Philhower's core DOES work with u8g2, at least in some configurations, though there are some issues (e.g. https://github.com/olikraus/u8g2/issues/2289). I don't know about using the MBED core. I will say that most of the community considers Earle Philhower's core the superior one.

You can ALSO use u8g2 without ANY arduino core, just using the basic Pico SDK (the chip's foundational library), but you have to write some interface code which as far as I know has not been integrated into the u8g2 distribution (I bet @olikraus would take a PR!), see https://github.com/olikraus/u8g2/issues/2159 for details.