olikraus/u8g2 #2324: Image gets inverted after some time becomes mirror image

sudheerbussa, 2023-12-14:
Hi i am using following 3 line code to print image on my GLCD. But it gets inverted after some time. am i missing any settings in constructor or any suggestions? ` U8G2_ST7565_ERC12864_ALT_F_6800 u8g2(U8G2_R0, 18, 17, 16, 15, 7, 6, 5, 4, 20, 8, 19, 9);// constructor
[image]
[image]
u8g2.clearBuffer();
  u8g2.drawXBMP(2, 2, INNOTRONIX_LOGO_width, INNOTRONIX_LOGO_height, INNOTRONIX_LOGO_bits);
  u8g2.sendBuffer();`

olikraus, 2023-12-14:
Looks like a hardware issue, where the display config memory has a bit change.
Maybe the problem goes away if you add a capacitor (1uF or so) close to the display power supply.

sudheerbussa, 2023-12-23:
I tried hardware changes. But not able to resolve problem. This problem only with **drawXBMP**. when i am using **drawStr** or print, there is no issue. But once i use **drawXBMP** and if my drawStr gets executed after the image inverted or mirrored, then my characters also mirrored or inverted as my drawStr is taking new mirrored display addresses. any other ways to display image instead of **drawXBMP** ?

olikraus, 2023-12-23:
hmm... I have no idea how this could happen.
Can you paste the complete code to reproduce this?