olikraus/u8g2 #2325: LCD not running - Sinda LCD- ST7565 - HELP!!! HELP!!! HELP!!!

Ashwini8935, 2023-12-15:
Hi,

I am seeking help for running a COG LCD (Datasheet Attached of the LCD). As per the datasheet, LCD has ST7567 Driver IC. Somewhere i am also confusing in the hardware interfacing with arduino. I made connection as follows:

1. GPIO 8 of Arduino UNO is connected with Pin no 3 (A0) of LCD.
2. GPIO 9 of Arduino UNO is connected with Pin no 2 (RES) of LCD.
3. GPIO 10 of Arduino UNO is connected with Pin no 1 (CS) of LCD.
4. GPIO 13 (SCK) of Arduino UNO is connected with Pin no 12 (D6:SCL) of LCD.
5. GPIO 11 (MOSI) of Arduino UNO is connected with Pin no 13 (D7:SI) of LCD.
6. Pin 14 of LCD - 3.0 V
7. pin 15 of LCD - GND
8. Pin 28 of LCD - GND

[image]

[PrintHelloWorld.zip](https://github.com/olikraus/u8g2/files/13686188/PrintHelloWorld.zip)

[128X64 (S) COG WHITE LCD Display-LC-553-D (1).pdf](https://github.com/olikraus/u8g2/files/13686128/128X64.S.COG.WHITE.LCD.Display-LC-553-D.1.pdf)

olikraus, 2023-12-15:
You need to use your pin numbers in the constructor

olikraus, 2023-12-15:
You must also add the required caps to the LCD and you also need to configure the bus mode for your display: see your datasheet

Ashwini8935, 2023-12-16:
Hi olikraus,

Thanks for your reply.

It will be really good if you let me know how to configure the bus mode of display..

olikraus, 2023-12-16:
It is described in the datasheet you provided.

Ashwini8935, 2023-12-16:
Hi,

I have used this diagram and code but the display is still not running.Please help

[hello_world_st7565.zip](https://github.com/olikraus/u8g2/files/13693207/hello_world_st7565.zip)

[image]

olikraus, 2023-12-17:
Oh, nice diagram. Looks correct from my perspective.

To be on the safe side, I suggest to connect c86 pin to GND.

 I also suggest to add the Arduino pin numbers into your diagram.

Once Arduino pin numbers are defined, we can derive the u8g2 constructor arguments.

Ashwini8935, 2023-12-18:
Hi,

Arduino pins are defined in the diagram if you can see properly and same pins have been used for constructor. For detailing i am attaching another diagram here. We are using **arduino UNO Board**

[image]

Ashwini8935, 2023-12-18:
Hi,

I request you to kindly share any whatsapp, phone no or email id for continuous communication without any delay.

olikraus, 2023-12-18:
> I request you to kindly share any whatsapp, phone no or email id

Why shell I do so? why shell I even respond to this issue? 
I have developed u8g2 in my spare time and i answer questions in my spare time. 
Remember: I do not earn a single doller while sitting and typing an answer here.

so... Take it or leave it... have fun...

Ashwini8935, 2023-12-18:
I am facing problem while running this and despite so many iterations i have not succeeded yet... The only guy who can help me into this is you. that's why i am asking it

Ashwini8935, 2023-12-18:
i have submitted diagram and code if you can help me. i am highly obliged

olikraus, 2023-12-18:
I see at least one more person who can help: Thats you.

Anyhow, you might have destroyed your display already, because there is no 5v to 3v3 level shifter.

Suggestion: (A) Use SW SPI until you see some usefull data on the LCD. (B) Run the U8g2 ContastTest to check whether your LCD requries a different contrast value. (C) Add a level shifter.

Ashwini8935, 2023-12-18:
Hi,

I might destroyed this LCD.. I am purchasing new one but it will take some time to reach in my hand.. I will message you once i received LCD.

olikraus, 2023-12-18:
Often 3v3 devices survice a 5v connection. Using a level shifter is important.

Pjatruha, 2025-10-25:
Yes, I agree with you, the datasheet says only about 3.3 volts. But people on the forums write that the displays work when connected directly to arduino. Here are some links with a positive result of this connection.
ST7567
https://youtu.be/Rlj9mM6Hsd8?si=ThWfUgWBzihWAGWW
ST7565
https://youtu.be/YwYkXSt6HfA?si=uLnhIk5cOpy9WJ9J
Don't get me wrong, I didn't mean to say you were wrong. I wanted to say that people have a positive experience connecting the display to logic lines with a voltage of 5 volts.