olikraus/u8g2 #2328: Display corruption with ST7920 on Teensy 4.0 using HW SPI

Compizfox, 2023-12-18:
I am not sure if it is a bug in u8g2 or in Teensyduino's SPI library, but the ST7920's display shows a lot of corruption with hardware SPI (see video).

https://github.com/olikraus/u8g2/assets/7603719/45df3587-c263-448a-91b8-61cf73916dd9

I tried to reduce the SPI bus clock speed, as well as changing the SPI mode, but without success.

This issue does not exist with software SPI.

The exact same firmware and display combination works fine on the Teensy LC, FWIW.

olikraus, 2023-12-18:
Now doubt, the ST7920 is the most sensitive display I know. 
Still the SPI mode might be an issue. Very long wires might also cause issues.

Compizfox, 2023-12-18:
I see. I tried all the SPI modes (0, 1, 2, 3), and the 'nature' of the corruption changed, but it didn't work properly with any of the modes.

The wires were the first thing I checked. The connections are sound and the wires are only 10 cm long.

olikraus, 2023-12-18:
How did you change the SPI mode? 
The mode will be set by u8g2, so you need to modify the u8g2 for the corresponding constructor, for example here: https://github.com/olikraus/u8g2/blob/34be7c84883a3ffc927e04eda7a99c539da040ee/csrc/u8x8_d_st7920.c#L171

Compizfox, 2023-12-18:
That's exactly where I changed it ;)