olikraus/u8g2 #2329: u8g2, controller and display

ErikJensen1, 2023-12-18:
I have been trying to use your display driver.
I have tried 2 displays(PowerTip PE12864WRF-042HC1Q or Winstar WO12864H-TFH#) both using the ST7567 controller (4W_SPI) but I am not sure what Setup I should use.

I have found the following setup will work:
uint8_t TX_Data_PT[] = {PE12864_LCDBiasRatio_1_9,PE12864_SetADCNormal,PE12864_SetCommonOutputReverse,
PE12864_SetResistorRatio(0x04),PE12864_ElectronicVolMode,PE12864_SetElectronicVolRegister(0x20),
PE12864_SetPowerControl(0x07),PE12864_SetStartLine(0x00),PE12864_SetPageAddress(0x00),
PE12864_SetColumnAddressUpper(0x00),PE12864_SetColumnAddressLower(0x00),PE12864_DisplayON,
PE12864_DisplayAllPointsOFF,PE12864_SetRAMModeNormal};
uint8_t TX_Data_WS[] = {PE12864_LCDBiasRatio_1_9,PE12864_SetADCNormal,PE12864_SetCommonOutputReverse,
PE12864_SetResistorRatio(0x05),PE12864_ElectronicVolMode,PE12864_SetElectronicVolRegister(0x28),
PE12864_SetPowerControl(0x07),PE12864_SetStartLine(0x00),PE12864_SetPageAddress(0x00),
PE12864_SetColumnAddressUpper(0x00),PE12864_SetColumnAddressLower(0x00),PE12864_DisplayON,
PE12864_DisplayAllPointsOFF,PE12864_SetRAMModeNormal};

But how do I find the Setup that fits these 2 displays?

Best regards
Erik Jensen

olikraus, 2023-12-18:
I suggest to try out some of the construtors mentioned here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7567-pi_132x64
Try also some constructors, which do not match exactly the dimensions of your display (but of course those with 128x64 should be prefered).
Also note, that your display might require a different contrast (so maybe you want to use the ContrastTest example)

Once you find a u8g2 constructor, which fits more or less good, then you could use the u8g2.sendF command to send your display specific settings to the display.

ErikJensen1, 2023-12-18:
I found a example "u8g2_template_stm32f103c8t6-master.zip" and used this to create my project.
First time it was compiled without any errors.
But it did not work. 
It used a software SPI.

So now I changed to the latest version 2.35.x,   

But what should I write in the cs, dc.
Is it only for Arduino?

ErikJensen1, 2023-12-18:
Now I used this example as template "[STM32_HAL_U8G2_OLED](https://github.com/eziya/STM32_HAL_U8G2_OLED/tree/master)"

I got some pixels on the display, but it looks strange.

Below is the code I copied and changed. Does this look right for the latest u8g2 version 2.35?

BR
Erik

uint8_t u8x8_stm32_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
/* STM32 supports HW SPI, Remove unused cases like U8X8_MSG_DELAY_XXX & U8X8_MSG_GPIO_XXX */
switch(msg)
{
case U8X8_MSG_GPIO_AND_DELAY_INIT:
/* Insert codes for initialization */
break;
case U8X8_MSG_DELAY_MILLI:
/* ms Delay */
HAL_Delay(arg_int);
break;
case U8X8_MSG_GPIO_CS:
/* Insert codes for SS pin control */
HAL_GPIO_WritePin(LCD_CSB_GPIO_Port, LCD_CSB_Pin, arg_int);
break;
case U8X8_MSG_GPIO_DC:
/* Insert codes for DC pin control */
HAL_GPIO_WritePin(LCD_A0_GPIO_Port, LCD_A0_Pin, arg_int);
break;
case U8X8_MSG_GPIO_RESET:
/* Insert codes for RST pin control */
HAL_GPIO_WritePin(LCD_RSTB_GPIO_Port, LCD_RSTB_Pin, arg_int);
break;
}
return 1;
}

uint8_t u8x8_byte_stm32_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
switch(msg) {
case U8X8_MSG_BYTE_SEND:
/* Insert codes to transmit data */
if(HAL_SPI_Transmit(&hspi3, arg_ptr, arg_int, TX_TIMEOUT) != HAL_OK) return 0;
break;
case U8X8_MSG_BYTE_INIT:
/* Insert codes to begin SPI transmission */
break;
case U8X8_MSG_BYTE_SET_DC:
/* Control DC pin, U8X8_MSG_GPIO_DC will be called */
u8x8_gpio_SetDC(u8x8, arg_int);
break;
case U8X8_MSG_BYTE_START_TRANSFER:
/* Select slave, U8X8_MSG_GPIO_CS will be called */
u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
HAL_Delay(1);
break;
case U8X8_MSG_BYTE_END_TRANSFER:
HAL_Delay(1);
/* Insert codes to end SPI transmission */
u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
break;
default:
return 0;
}
return 1;
}

u8g2_Setup_st7567_enh_dg128064_1(&u8g2, U8G2_R0, u8x8_byte_stm32_hw_spi, u8x8_stm32_gpio_and_delay);
u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
u8g2_SetPowerSave(&u8g2, 0); // wake up display

olikraus, 2023-12-18:
Difficult... This project (U8g2) is an Arduino Library, which can be installed via the Arduino IDE. 
My suggestion still is the same, try to use different setup procedures for your controller, see here: https://github.com/olikraus/u8g2/wiki/u8g2setupc

ErikJensen1, 2023-12-19:
When you read the "https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform" 

[image]

I get the impression that it can only work with the SW spi. Is this correct?

olikraus, 2023-12-19:
> I get the impression that it can only work with the SW spi. Is this correct?

U8g2 will work with hardware SPI, but you have to tell U8g2 how to do this. For this purpose your have to create and provide a callback function, which will do the hardware SPI transfer.

ErikJensen1, 2024-01-05:
How do  "u8g2_Setup...()" actually see that you are using a sw or hw spi? Will it send all signals? Both for sw_spi and hw_spi?
hw_spi U8X8_MSG_BYTE_SEND,....
sw_spi U8X8_MSG_GPIO_SPI_CLOCK, U8X8_MSG_GPIO_SPI_DATA

ErikJensen1, 2024-01-05:
I think I am on the right track now. But I get this error: "../Drivers/u8g2/cppsrc/U8x8lib.h:157:8: error: expected identifier or '(' before string constant"
[image]
I have an idea it is from the constructor:
U8G2_ST7567_PI_132X64_1_4W_HW_SPI u8g2(&u8g2, ....);
or missing compiler #define ....
But where can I see this?

ErikJensen1, 2024-01-05:
The display is now changing :-) between blank display, all pixels low contrast or all pixels black.
We try to write "Hello world" with a font but this can not be seen in any of the 3 cases.
  u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
  u8g2_DrawStr(&u8g2, 10, 15, "Hello World!");
Any suggestion on what can be wrong are welcome.

The setup for the 3 cases.
  //Blank display:
  //u8g2_Setup_st7567_jlx12864_1(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_stm32_gpio_and_delay);
  //All pixels low contrast:
  //u8g2_Setup_st7567_enh_dg128064_1(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_stm32_gpio_and_delay);
  //All pixels black:
  u8g2_Setup_st7567_os12864_1(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_stm32_gpio_and_delay);

olikraus, 2024-01-05:
> Any suggestion on what can be wrong are welcome.

You need to send the buffer to the display with the picture loop or the sendBuffer command.
The sendBuffer additionally requires, that you use the "F" constructor: u8g2_Setup_st7567_jlx12864_f

closed 2024-01-08 by ErikJensen1