olikraus/u8g2 #2330: Read - Write to SD Card

CWGwaltney, 2023-12-19:
Hello... I am new and not fully understanding how the image buffer is accessed in code
I am wanting to send the display to another setup over bluetooth
I understand i could just send the commands over bluetooth and have the second setup render the image
i want to send the image, i understand this may be slow

I was wondering if you could add to Load from SD sketch to add Save to SD

This would be useful, like if a tic-tac-toe board game, draw the play field and save to SD and then instead of 
redrawing the vectors, just load the raster on game reset for example, kind-of-like a background
(i understand this would probably be much slower, just wondering if it is possable)
also a snapshot of the display state could be saved, like for data logging an event
screenshot

the Save to SD will save the current image buffer to SD (instead of just the logo image)

I understand that the process is to read a bite and send a bite, but i am not fully understanding
why is it not a 2D array with x and y ?, or more specific, row and column
i understand the concept of pages and tiles not the same as pixels

also i am wondering if A2Printer can be revised to possably print to bluetooth, serial, IIC
can a communications port be added like in setup constructor?
i am assuming this is a bit banging type of thing

also i am not understanding how directaccess sketch is working 

i am not asking for all the work be done, but if you could add some explination, tutorial or point to a link
needless to say i have googled and not found a good explination

thanks! CWG

olikraus, 2023-12-19:
Which constructor do you use? The F constructor might be more simpler here. With the F constructor you would do something like this:

``` C++
uint16_t cnt = u8g2.getBufferSize();
uint8_t *ptr = u8g2.getBufferPtr();
uint16_t i;
for( i = 0; i < cnt; i++ )
  send_byte(*ptr++);
```

CWGwaltney, 2023-12-19:
thanks, i am investigating the F constructor

CWGwaltney, 2023-12-23:
ok i understand the F constructor, i found the WritetoSD.ino and ScreenShot.ino and A2Printer

I am still not exactly understanding what is going on, but i see now how screenshot can print to the serial monitor, how can i get it to print to sd card?  how can i print to bluetooth for example?  thanks

olikraus, 2023-12-23:
Writing to SD card or Bluetooth is beyond the scope of u8g2, so maybe it is better to ask this in Arduino Forum.