olikraus/u8g2 #2331: Issues with GMG12864 - Please help.

navado, 2023-12-21:
Hello
This started as a discussion in discussions section, but it seems none read those.

I'm trying to bring up ESP32 with GMG12864 that I have in the spares and encountering several questions using u8g2.
# My setup:
Display: GMG12864-06D
```C++

#define CS_PIN PC14
#define RST_PIN PC15
#define RS_PIN PA0
#define SCL_PIN PA5
#define SDO_PIN PA6
#define SDI_PIN PA7
#define BACKLIGHT_PIN PinName::PA_1
#define LED_BRD PC13

#define CONTRAST_SETTING 0x3A

U8G2_ST7565_EA_DOGM128_1_4W_SW_SPI lcd(U8G2_R2,
                                       /* clock=*/SCL_PIN,
                                       /* data=*/SDO_PIN,
                                       /* cs=*/CS_PIN,
                                       /* dc=*/RS_PIN,
                                       /* reset=*/RST_PIN
                                       );

```
__This works__
Now the questions
## 1. How to initialize the display with HW SPI?
I dug into different implementations of constructors initializing ST7565 with HW SPI and it looks like they're all referring to hard coded pin numbers that are not corresponding to BluePill pins. Probably I'm missing something, but also I'm unable to find a working config with HW SPI

## 2. What's the dc pin in the constructor?
I just tried the RS pin from the display and it worked, but why?

## 3. How to configure offset for the display?
I'm getting 125x64 useful pixels shifted 4 pixels right if initializing with `U8G2_R2` and shift of 8 pixels if initialized with `U8G2_R0` giving it 121x64 useful pixels.

Appreciate clues to the right direction.

[image]

olikraus, 2023-12-23:
> I'm getting 125x64 useful pixels shifted 4 pixels right

Please try some other constructors. All are listed here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-ea_dogm128

Ignore the contrast, you can adjust the contrast with setContrast() with any of the constructors.

navado, 2023-12-25:
I tried, no luck. However with another SW constructor `U8G2_ST7565_ERC12864_ALT_F_4W_SW_SPI` I got full screen working, i.e. issue number 3 solved

```C++

/* ------------------- THESE WORKING -----------------------*/
// U8G2_ST7565_EA_DOGM128_1_4W_SW_SPI lcd(U8G2_R2,
//                                        /* clock=*/SCL_PIN,
//                                        /* data=*/SDO_PIN,
//                                        /* cs=*/CS_PIN,
//                                        /* dc=*/RS_PIN,
//                                        /* reset=*/RST_PIN
//                                        );

// U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI lcd(U8G2_R2,
//                                         /* clock=*/SCL_PIN,
//                                         /* data=*/SDO_PIN,
//                                         /* cs=*/CS_PIN,
//                                         /* dc=*/RS_PIN,
//                                         /* reset=*/RST_PIN
//                                         );

// Uses full screen
U8G2_ST7565_ERC12864_ALT_F_4W_SW_SPI lcd(U8G2_R0,
                                        /* clock=*/SCL_PIN,
                                        /* data=*/SDO_PIN,
                                        /* cs=*/CS_PIN,
                                        /* dc=*/RS_PIN,
                                        /* reset=*/RST_PIN
                                        );

/* ------------------- THESE NOT WORKING -----------------------*/

// U8G2_ST7565_ERC12864_ALT_F_4W_HW_SPI lcd(U8G2_R0,
//                                         /* cs=*/CS_PIN,
//                                         /* dc=*/RS_PIN,
//                                         /* reset=*/RST_PIN
//                                         );

// U8G2_ST7565_EA_DOGM128_1_4W_HW_SPI lcd(U8G2_R2,
//                                         /* cs=*/CS_PIN,
//                                         /* dc=*/U8X8_PIN_NONE
//                                         );

// U8G2_ST7565_EA_DOGM128_2_4W_HW_SPI lcd(U8G2_R2,
//                                         /* cs=*/CS_PIN,
//                                         /* dc=*/RS_PIN,
//                                         /* reset=*/RST_PIN
//                                         );
// U8G2_ST7565_EA_DOGM128_1_2ND_4W_HW_SPI lcd(U8G2_R2,
//                                         /* cs=*/CS_PIN,
//                                         /* dc=*/RS_PIN,
//                                         /* reset=*/RST_PIN
//                                         );
// U8G2_ST7565_EA_DOGM128_1_3W_HW_SPI lcd(U8G2_R2,
//                                         /* cs=*/CS_PIN,
//                                         /* reset=*/RST_PIN
//                                         );

```

olikraus, 2023-12-26:
Point 1:
HW SPI depends on your board. U8g2 will just use the Arduino SPI object.
Pin numbers are given here: https://www.arduino.cc/reference/en/language/functions/communication/spi/
Of course for any none-Arduino boards you need to consult the documentation of your board regarding the HW pin numbers.

Point 2:
RS, DC, CD, A0... are names from the datasheet. They are not consistent. In u8g2 it is named "dc" (data/command, see also here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#wiring)

navado, 2023-12-26:
Thank you!
This helps.

closed 2024-01-30 by navado