olikraus/u8g2 #2332: 96x65 display support

MagicNeo75, 2023-12-21:
Hi, 
I´m working with TM9665ACCWGWC4 wich is a 96x65 pixels LCD.
https://es.aliexpress.com/i/32683763108.html?gatewayAdapt=glo2esp
I have it connected to an ESP32-WROOM running my FW made in VScode and platformio.
I`ve tried to use your library with some ST7567 and ST7565 and the result is alwas the same: only 64 colums are written and the number 65 stays with some pixles ON and others OFF. Like this:
[image]

Is there any change I can make to solve this behaviour and show at least the 65th column always OFF?

Thanks in advance.

olikraus, 2023-12-23:
which costructor did you use for the above image?

MagicNeo75, 2023-12-23:
This is the class:

U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_R3, /* cs=*/ 19, /* dc=*/ 21, /* reset=*/ 18);  

this the init function:

void display_init(void)
{
  SPI.begin(23,16,22);
  u8g2.begin();
  u8g2.setContrast(115);
}

and this is the screen constructor:

void draw_menu_main(uint8_t select)
{
    u8g2.setDrawColor(1);
    u8g2.drawBox(0, 32, 64, 96);

    u8g2.setFont(u8g2_font_tiny_simon_mr);
    u8g2.setFontDirection(0);
    u8g2.setFontMode(0);
    u8g2.setDrawColor(0);
    u8g2.drawStr(2,39,"MENU");
    u8g2.drawHLine(25, 34, 64);
    u8g2.drawHLine(25, 35, 64);
    u8g2.drawHLine(25, 37, 64);

    u8g2.setDrawColor(0);
    u8g2.drawBox(0, 40, 64, 88);
    u8g2.setFont(u8g2_font_unifont_h_symbols);

    if(select == 0)
    {
      u8g2.setDrawColor(1);
//      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-1, 62, MENU_BOX_HEIGHT+2);
      u8g2.drawBox(1, 40, 62, ((MENU_BOX_HEIGHT)*2)+1);
      u8g2.setDrawColor(0);
      u8g2.drawBox(2, 41, 60, ((MENU_BOX_HEIGHT)*2)+1);
      u8g2.setDrawColor(1);
      u8g2.drawStr(5,41+(2*(MENU_BOX_HEIGHT-1))-2,"CONFIG.");
      u8g2.drawStr(20,41+(MENU_BOX_HEIGHT-1)-1,"VER");
      u8g2.drawHLine(1, 39, 62);
      u8g2.drawHLine(1, 40, 62);
    }
    else
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, 41, 62, ((MENU_BOX_HEIGHT-1)*2)+1);
      u8g2.setDrawColor(0);
      u8g2.drawStr(5,41+(2*(MENU_BOX_HEIGHT-1))-2,"CONFIG.");
      u8g2.drawStr(20,41+(MENU_BOX_HEIGHT-1)-1,"VER");
      u8g2.drawHLine(1, 40, 62);
    }

    if(select == 1)
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-1, 62, MENU_BOX_HEIGHT+2);
      u8g2.setDrawColor(0);
      u8g2.drawBox(2, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1)), 60, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(1);
      u8g2.drawStr(5,(MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"PROGRA.");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-2, 62);
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-1, 62);
    }
    else
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawStr(5,(MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"PROGRA.");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-1, 62);
      if(select == 0)
      {
        u8g2.setDrawColor(1);
        u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(1*(MENU_BOX_HEIGHT+1))-2, 62);
      }
    }

    if(select == 2)
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1))-1, 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawBox(2, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1)), 60, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(1);
      u8g2.drawStr(15,(MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"TEST");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1))-2, 62);
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1))-1, 62);
    }
    else
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawStr(15,(MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"TEST");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1))-1, 62);
      if(select == 1)
      {
        u8g2.setDrawColor(1);
        u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(2*(MENU_BOX_HEIGHT+1))-2, 62);
      }
    }

    if(select == 3)
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawBox(2, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1)), 60, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(1);
      u8g2.drawStr(3,(MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"MOD.COD");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1))-2, 62);
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1))-1, 62);
    }
    else
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawStr(3,(MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"MOD.COD");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1))-1, 62);
      if(select == 2)
      {
        u8g2.setDrawColor(1);
        u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(3*(MENU_BOX_HEIGHT+1))-2, 62);
      }
    }

    if(select == 4)
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawBox(2, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)), 60, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(1);
      u8g2.drawStr(12,(MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"SALIR");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1))-2, 62);
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1))-1, 62);
      u8g2.drawHLine(1, (MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-1, 62);
    }
    else
    {
      u8g2.setDrawColor(1);
      u8g2.drawBox(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)), 62, MENU_BOX_HEIGHT);
      u8g2.setDrawColor(0);
      u8g2.drawStr(12,(MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1)))+MENU_BOX_HEIGHT-2,"SALIR");
      u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1))-1, 62);
      if(select == 3)
      {
        u8g2.setDrawColor(1);
        u8g2.drawHLine(1, MENU_BOX_LINE_BASE+(4*(MENU_BOX_HEIGHT+1))-2, 62);
      }
    }
}

with select = 1

You can see as I start at column 0, but theres in an aditional line (the 65th) in the position -1.
I've tried to use 180 grades rotation and the aditional line is in the same position.

olikraus, 2023-12-26:
I have created a new beta release 2.35.9. It contains the new constructor `U8G2_ST7567_96X65_F_4W_HW_SPI`.
There might be still some issues with the new constructor. Especially there still might be a wrong offset.

Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

#

MagicNeo75, 2024-01-18:
Hi, Sorry for my late reply.

I'm using your updated library and now it works perfecly!!.

Thank you very much!

closed 2024-08-31 by olikraus