olikraus/u8g2 #2334: Intereting SSD1306 vs SH1107 - Seeed Display

CWGwaltney, 2023-12-25:
products,

https://www.seeedstudio.com/Seeeduino-XIAO-Expansion-board-p-4746.html

https://www.seeedstudio.com/Grove-OLED-Display-0-96.html
[image]

https://www.seeedstudio.com/Grove-OLED-Display-1-12-SH1107-V3-0-p-5011.html
[image]

Note: 
Three OLED Display from Seeed Studio
Base U8g2Logo.ino
with 

// U8g2 Contructor List (Picture Loop Page Buffer)
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1107_SEEED_128X128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
// End of constructor list

[image]

initial run with SSD1306 two displays correct, SH1107 upside down and shifted with random data below row 64
Note, all displays with 0 rotation and display flex cable facing down
understandable as SSD1306 is 128X64

[image]

without powering down display, upload code revised to SH1107

SH1107 display correct, two SSD1306 displays upside down and image is progressive scan

https://github.com/olikraus/u8g2/assets/154296902/5ff558db-0bc9-48ff-9097-f6491d6479f9

then without power off, reload initial SSD setup and run two display right side up, one upside down

[image]

[image]

Also SSD is HW_I2C and SH1107 only has SW_I2C

would ask that the contructor for Seeed SSD1306 128X64 have an option for 128X128 (only 64 will be viewable)
and SH1107 128X128 so that one constructor can drive both displays, and only the 64 will be truncated

olikraus, 2023-12-26:
I am confused what to do here. 
In general one constructor can only drive one display.

CWGwaltney, 2023-12-27:
thanks, i am going to investigate further... ill report back

CWGwaltney, 2023-12-27:
FYI

Inventory of Seeed OLED Display devices
***************************************************************************************************************************
Grove - OLED Display 0.66" (SSD1306)- IIC - 3.3V/5V SKU 104020248
Based on SSD1306, cropped to 64*48 resolution(Monochrome)
Changeable I2C address 0X7A & 0X78
[image] [image]

***************************************************************************************************************************
Grove - OLED Display 0.96" (SSD1306, SSD1308) upgraded to (SSD1315) I2C - 3.3V/5V Interface SKU 104020208 V1.0
Based on SSD1315, cropped to 128*64 resolution(Monochrome)
Changeable I2C address 0X7A & 0X78
[image] [image]
*****************************************************************************************************************************
Seeed Studio Expansion Board Base for XIAO with Grove OLED (SSD1306) - IIC SKU 103030356 V1.0
Based on SSD1306, cropped to 128*64 resolution(Monochrome)
Not Changeable I2C address 0X7A 
[image]
*****************************************************************************************************************************
Grove - OLED Display 0.96" upgraded to (SSD1308Z) I2C - 3.3V/5V Interface SKU 104020208 V1.1
Based on SSD1308Z, cropped to 128*64 resolution(Monochrome)
Changeable I2C address 0X7A & 0X78

[image]

Grove - OLED Yellow&Blue Display 0.96 (SSD1315) - SPI/IIC -3.3V/5VSKU 104020249
Based on SSD1315, cropped to 128*64 resolution(Monochrome)
Changeable SPI TO IIC
Changeable I2C address 0X7A & 0X78
[image] ![image (https://github.com/olikraus/u8g2/assets/154296902/459138f8-e58b-4a01-93ef-21e68a3ffd1e)
***************************************************************************************************************************
Grove - OLED Display 1.12" (SSD1327) I2C - 3.3V/5V SKU 104030011 V1.0
96×96 pixel dot matrix Supports Continuous Horizontal Scrolling 16 color grayscale 
OLED module ([ LY120])and SSD1327 driver IC.
[image] 
[image]
***************************************************************************************************************************
Grove - OLED Display 1.12" (SH1107G) I2C - 3.3V/5V SKU 101020452 V2.0
128×128 pixel dot matrix Supports Continuous Horizontal Scrolling (Monochrome)
OLED module ([ LY120])and SH1107G driver IC.
***************************************************************************************************************************
Grove - OLED Display 1.12 (SH1107) V3.0 - SPI/IIC -3.3V/5V SKU 104020250 V3.0
128×128 pixel dot matrix Supports Continuous Horizontal Scrolling (Monochrome)
OLED module ([ LY120])and SH1107 driver IC.
Changeable SPI TO IIC
Changeable I2C address 0X3C & 0X3D
[image] [image]

CWGwaltney, 2023-12-27:
I think I found the problem, it i hardware and not fixable in software
the 128x64 displays the chip & ribbon cable attaches to the front/top of the board
the 128x128 display the chip & ribbon cable attache to the back/bottom of the board

this results in a mirror image of the chip and ribbon cable pinout from left to right 
resulting in a hardware mirroring on the device

CWGwaltney, 2023-12-27:
I tried to make a custom contructor using U8X8_SH1107_SEEED_128X128_HW_I2C and trying to make it 128X64
and got this error: 'u8x8_d_sh1107_seeed_128x64' was not declared in this scope

any ideas?

In file included from c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:53:0,
                 from C:\Users\cgwal\Documents\Arduino\XIAO-U8g2\examples\full_buffer\U8g2Logo\U8g2Logo.ino:6:
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8x8lib.h: In constructor 'U8X8_SH1107_SEEED_128X64_HW_I2C::U8X8_SH1107_SEEED_128X64_HW_I2C(uint8_t, uint8_t, uint8_t)':
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8x8lib.h:1869:27: error: 'u8x8_d_sh1107_seeed_128x64' was not declared in this scope
     u8x8_Setup(getU8x8(), u8x8_d_sh1107_seeed_128x64, u8x8_cad_ssd13xx_fast_i2c, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8x8lib.h:1869:27: note: suggested alternative: 'u8x8_d_sh1107_seeed_128x128'
     u8x8_Setup(getU8x8(), u8x8_d_sh1107_seeed_128x64, u8x8_cad_ssd13xx_fast_i2c, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
                           u8x8_d_sh1107_seeed_128x128
In file included from C:\Users\cgwal\Documents\Arduino\XIAO-U8g2\examples\full_buffer\U8g2Logo\U8g2Logo.ino:6:0:
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8g2lib.h: In constructor 'U8G2_SH1107_SEEED_128X64_F_HW_I2C::U8G2_SH1107_SEEED_128X64_F_HW_I2C(const u8g2_cb_t*, uint8_t, uint8_t, uint8_t)':
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:4791:5: error: 'u8g2_Setup_sh1107_i2c_seeed_128x64_f' was not declared in this scope
     u8g2_Setup_sh1107_i2c_seeed_128x64_f(&u8g2, rotation, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\cgwal\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:4791:5: note: suggested alternative: 'u8g2_Setup_sh1107_i2c_seeed_128x128_f'
     u8g2_Setup_sh1107_i2c_seeed_128x64_f(&u8g2, rotation, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     u8g2_Setup_sh1107_i2c_seeed_128x128_f
Multiple libraries were found for "Adafruit_ZeroDMA.h"
  Used: C:\Users\cgwal\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.5\libraries\Adafruit_ZeroDMA
  Not used: C:\Users\cgwal\Documents\Arduino\libraries\Adafruit_Zero_DMA_Library
exit status 1

Compilation error: exit status 1

CWGwaltney, 2023-12-29:
OK i have found the two constructors who operate each display

// U8g2 Constructor List (Frame Buffer)
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SH1107_SEEED_128X128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// End of constructor list

how do i set them up to share the same frame buffer 128x128 and send 128x64 to one and 128x128 to the other

fyi i am using the standard logo sketch

olikraus, 2023-12-30:
> how do i set them up to share the same frame buffer 128x128 and send 128x64 to one and 128x128 to the other

This is not forseen. However you can create the a draw() procedure and call it twice with two different u8g2 objects.
something like this:
``` C++
void draw( U8G2 &u8g2 )
{
  u8g2.setFont(...);
  u8g2.drawStr(...)
} 
```

U8G2 is the base class for u8g2 objects.