olikraus/u8g2 #2337: STM32F407G-DISC1 on LCD 128x64 WG12864

DustyTR, 2023-12-27:
I am trying to use my 128x64 Graphical **WG12864** LCD on STM32F407G, also I am using Platform/IO to program it, but there is a problem library is working but I can't get anything on my screen. Maybe there is a problem with wiring but I checked it and there is nothing wrong I think. Does anybody have ideas that I can try?

Photo of my project

[image]

Connections:

[image]

**platformio.ini**

```
[env:black_f407vg]
platform = ststm32
board = black_f407vg
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
lib_deps = 
	olikraus/U8g2@^2.35.8
	thomasfredericks/Bounce2@^2.71

```

C++ code:

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <MUIU8g2.h>
#include <Bounce2.h>
#include <Wire.h>

U8G2_SH1106_128X64_WINSTAR_1_6800 u8g2(U8G2_R0, PC_15, PH_1, PE_2, PE_4, PB_8, PE_0, PE_6, PC_14, /*enable=*/ PC_13 /* A4 */, /*cs=*/ U8X8_PIN_NONE, /*dc/rs=*/ PE_3 /* A3 */, /*reset=*/ PH_0 /* A1 */);
//U8G2_SH1106_128X64_WINSTAR_1_6800 u8g2(U8G2_R0, PE_3, PE_0, PB_8, PE_4, PE_2, PH_1, PC_15, PC_13, PE_5, U8X8_PIN_NONE, PB_9, PH_0);
//U8G2_SH1106_128X64_WINSTAR_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PC13, /* data=*/ PB7, /* cs=*/ PB5, /* dc=*/ PB9, /* reset=*/ PH_0);

void setup(void) {
    u8g2.begin();
    u8g2.setFont(u8g2_font_ncenB08_tr);
}

void loop(void) {
    u8g2.clearBuffer();
    u8g2.setCursor(20, 20);
    u8g2.print("Hello, World!");
    u8g2.sendBuffer();

    delay(2000);
}
```

Maybe this can help:

> https://www.winstar.com.tw/products/graphic-lcd-display-module/wg12864c.html

Thanks for your interest.

DustyTR, 2024-01-02:
Update and please someone can help?

I used HW-221 for the LCD Screen because I noticed the STM32F407G-DISC1's microcontroller is working on 3V, and it also outputs as 3V. So, I tried a logic-level converter for the 'Data bus line's. I also noticed that I am using `[env:black_f407vg]` I changed it to `[env:disco_f407vg]` for my STM32F407G-DISC1.

I used an optocoupler **TLP521-1** for other connections (D/I, R/W, Enable, and Reset) that will work like a logic-level converter.

I hope this helps.

**C++ code:**

```
#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>

#define cs1 PB5
#define cs2 PB7

#define d0 PB8
#define d1 PE0
#define d2 PE2
#define d3 PE4
#define d4 PE6
#define d5 PE3
#define d6 PE5
#define d7 PC13

#define enable PE1
#define di PC14
#define rw PB9 

#define reset PC15

U8G2_KS0108_128X64_1 u8g2(U8G2_R0, d0, d1, d2, d3, d4, d5, d6, d7, enable, di, rw, cs1, cs2, reset); 

void buzzer(int time) {
    digitalWrite(PB4, HIGH);
    delay(time);
    digitalWrite(PB4, LOW);
}

void setup(void) {
    pinMode(PB4, OUTPUT);
    u8g2.begin();
    buzzer(1000);
}

void loop(void) {
    u8g2.clearBuffer();

    u8g2.setFont(u8g2_font_ncenB08_tr);
    u8g2.drawStr(0, 10, "Hello, World!");

    u8g2.sendBuffer();

    delay(500);

    u8g2.clearBuffer();

    u8g2.sendBuffer();

    delay(500);
}
```

**platformio.ini**

```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = arduino
upload_protocol = stlink
lib_deps = 
	olikraus/U8g2@^2.35.8
	thomasfredericks/Bounce2@^2.71
```

[image]

olikraus, 2024-01-03:
To me your display seems to have a KS0108 controller, so you probably need to use a different constructor. Moreover you need to connect V0 via a varpot to vee.

Horta-git, 2024-01-08:
Yep this is a KS0108 controller, by the way I´m also trying to make it work with u8g2... any tiny bit of info is very welcome.

olikraus, 2024-01-08:
A tiny bit of info is the picture of my own KS0108 device: https://github.com/olikraus/u8g2/wiki/gallery#15-dec-2016-ks0108-128x64-lcd

Also note, that a 5V Arduino Board will nicely fit to the 5V KS0108 controller. A 3.3V Arduino board will require a level shifter.

closed 2024-06-02 by DustyTR