olikraus/u8g2 #2339: U8x8log Echo serial Print

CWGwaltney, 2023-12-29:
Sorry if this is a bothersome question,

Is there a way to get the serial print command to automatically also print to U8x8log, instead of having to construct an identical statement of syntax u8x8

 // Print a number on the U8g2log window
  u8g2log.print(millis());
// Print same number on the Serial Monitor
Serial.print(millis());

Thanks!

olikraus, 2023-12-30:
Idea:
1. Create a new class, derived from the Arduino Print class (https://www.arduino.cc/reference/en/language/functions/communication/print/)
2. Add u8g2log and serial as members
3. Overwrite the write() member function and write to u8g2log and Serial

Not sure, whether this will work, but it might be a starting point.

CWGwaltney, 2024-01-01:
OK I am not exactly understanding this, I will work on learning
Do you think you can add this as a feature update for U8X8LOG
some type of toggle possably... i mean it is supposed to act like a serial terminal

Thanks!

olikraus, 2024-01-05:
maybe I can create such an example once i have some time...

olikraus, 2024-09-01:
This seems to work:

``` C++
#define U8LOG_WIDTH 16
#define U8LOG_HEIGHT 8
uint8_t u8log_buffer[U8LOG_WIDTH*U8LOG_HEIGHT];
U8X8LOG u8x8log;

class extendedSerialLog : public Print
{
  size_t write(uint8_t v) 
  {
    Serial.write(v);
    u8x8log.write(v);
    return 1;
  }
};

extendedSerialLog xlog;

void setup(void)
{
  Serial.begin(9600);
  u8x8.begin();
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  
  u8x8log.begin(u8x8, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer);
  u8x8log.setRedrawMode(1);   // 0: Update screen with newline, 1: Update screen for every char  
}

void loop(void) {
  xlog.print("millis=");
  xlog.print(millis());
  xlog.print("\n");
  delay(500);
}

```

olikraus, 2024-09-01:
Maybe i misunderstood your request in https://github.com/olikraus/u8g2/issues/2454
Hope that the above code solves your problem.
I also added the example to the u8log docu.

closed 2024-09-01 by olikraus

tlmelk, 2024-12-07:
Will this only work for numbers? Anyway it could modified for text? I tried adding it into some code, but all my Serial.print() involve a text msg. Seem to not work with my code as is. Typical output are like:

      Serial.println("TLE set #:" + String(SAT));

Thanks in advance for your reply!