olikraus/u8g2 #2341: Question: what might be the problem here?

tilllt, 2024-01-02:
This is not so much an Issue with the library, more a question. I had a generic Aliexpress SH1106 SPI display hooked up to a Nano33 BLE and it ran the U8G2 examples fine. 

For some unrelated reason I needed to switch the platform to ESP32, the one I had laying around is a LilyGo ESP32_S2_WOOR (Wroom) - the minimal variant without built-in Display. Unfortunately the documentation of the Lilygo devices is ... Not great, but following some posts I found I finally managed to get an image.

But it's flipped, distorted etc.
https://youtu.be/_mbK70SZBJA

I am not knowledgeable enough to see if this is a 

- wiring mistake

- configuration error

- something else?

Would be great if you could take a look and see if you have an idea what could be wrong.

Thanks

olikraus, 2024-01-02:
Which constructor did you use?

tilllt, 2024-01-02:
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 34, /* dc=*/ 37, /* reset=*/ 38);

olikraus, 2024-01-02:
Did you try to reduce the bus speed with the corresponding u8g2 command?

tilllt, 2024-01-03:
Changing the clock speed to

`void setup(void) {
  u8g2.setBusClock(4000000);
  u8g2.begin();
}`

or 

`void setup(void) {
  u8g2.setBusClock(8000000);
  u8g2.begin();
}`

makes the distortion look different (worse in fact, nothing is legible anymore).

The default speed, works "best" it seems unfortunately not "good". What IS the default speed for the SH1106 with HW SPI?

olikraus, 2024-01-03:
Default/ max Speed should be mentioned by the display vendor

tilllt, 2024-01-03:
FYI the pololu sh1106 library works OOB, so i am more confident its not my wiring after all.

https://github.com/pololu/pololu-oled-arduino

tilllt, 2024-01-03:
Adafruit SH110x Library also works OOB

https://github.com/adafruit/Adafruit_SH110x

olikraus, 2024-01-05:
not sure what I can say... for me the SH1106  driver did work...

tilllt, 2024-01-06:
Well, with everything that I tried it might be safe to say that u8g2 driver for the sh1106 connected to an esp32-s2 doesn't work out of the box. 

I am not knowledgeable enough to further debug the reason for this but I can't think of any other mistake I could have made, so I am leaving this here as a bug report / comment and will try my luck with another library.

- u8g2 graphics demo works when connected to the Arduino Nano 33 BLE with this exact display. 

- This same display works when connected to a LilyGo ESP32-S2, using the Pololu SH1106 driver

- This same display works when connected to a LilyGo ESP32-S2, using the Adafruit SH1106 driver

- This same display DOES NOT WORK when connected to a LilyGo ESP32-S2, using the u8g2 with the same constructor that worked on the Arduino.

olikraus, 2024-01-06:
Maybe the default Arduino SPI pin assignment is still wrong. You could also try to use the u8g2 pin remapping feature.
``` C++
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 34, /* dc=*/ 37, /* reset=*/ 38, /* force clk= */ 36, /* force MOSI=*/ 35);
```

tilllt, 2024-01-07:
If i use the exact line you suggested i get 
```
C:\Users\xyz\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:2712:7: note:   candidate expects 1 argument, 6 provided

exit status 1

`Compilation error: no matching function for call to 'U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI::U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI(const u8g2_cb_t*, int, int, int, int, int)'
```

If i use

`U8G2_SH1106_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* cs=*/ 34, /* dc=*/ 37, /* reset=*/ 38, /* force clk= */ 36, /* force MOSI=*/ 35);`

it compiles but it doesnt produce anything on the display

olikraus, 2024-01-12:
hmm good point, my mistake. This remapping feature is not available with HW SPI.
Did you try SW SPI?
``` C++
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* clk= */ 36, /* MOSI=*/ 35, /* cs=*/ 34, /* dc=*/ 37, /* reset=*/ 38);
```

tilllt, 2024-01-12:
> Did you try SW SPI?

Yeah I tried both, no luck.

olikraus, 2024-01-19:
i am running out of ideas. If SW SPI doesn't work, then there probably is a wiring issue.

tilllt, 2024-01-21:
Just out of curiosity: what could be a wiring issue that makes the display work with other libraries but not u8g2?

olikraus, 2024-01-21:
> Just out of curiosity: what could be a wiring issue that makes the display work with other libraries but not u8g2?

good point. U8g2 requires Arduino Pin numbers, but if Adafruit GFX works, then also U8g2 should work.
Maybe you could post both, your working example with that other lib and your u8g2 example, which doesn't work.