olikraus/u8g2 #2344: Renaming LW12832 to YXD12832

shezik, 2024-01-11:
Hi olikraus,

6 months ago I opened a [pull request](https://github.com/olikraus/u8g2/pull/2217) to add support for a 12832 display named LW12832 and it was merged, thank you!

However I just figured out that the real name of the display is ***YXD-12832D02AX2*** and the current LW12832 name is rather uninformative. The real name is written on the ribbon cable and I never noticed. :(

Should this display get renamed in the codebase and introduce a breaking change or just make a note of it in the wiki?

v1993, 2024-01-11:
You can add a bunch of defines to maintain deprecated name in public interface if backward compatibility is vital. Definitely go for the correct name IMO and maybe note on wiki why the deprecated alias exists.

olikraus, 2024-01-11:
I can also create a new constructor with the desired name...

shezik, 2024-01-11:
> I can also create a new constructor with the desired name...

That would be great!

The LW thingy is the name abbreviation of the retailer that sold me this obscure display... you probably want to get rid of the old constructor ;)

olikraus, 2024-08-31:
added YXD12832 constructor

closed 2024-08-31 by olikraus