olikraus/u8g2 #2348: Pixel Loss

Alejandro13C, 2024-01-12:
Hi all,

I see how i lost a lot of pixel.
Some picture:
[image]
[image]
[image]

`u8g2.firstPage();
  SerialLogger();
  do {

    if(Vbat < 12.0){    //If the battery is less than 12.0, show battery voltage.
      //Converts from float to const char
      dtostrf(Vbat, 4, 2, Battery_char); 
      u8g2.drawStr(35 , 60, Battery_char);
      u8g2.drawStr(85 , 60 , "V");
      u8g2.drawXBMP(48, 2, 40, 40, BatteryLogo);

    } else {

      // Else show Oil pressure value.
      dtostrf(Oil_pressure, 4, 2, OP_char);
      u8g2.drawStr(75 , 60 , "Bar");
      u8g2.drawXBMP(48, 10, 32, 32, OilLogo);

    }
  } while ( u8g2.nextPage() );
  delay(1000);
`

Thanks in Advance!

olikraus, 2024-01-12:
Never change the content within the "while" first/next page loop.
To solve this, move the while loop into the if/else body.

Alejandro13C, 2024-01-17:
Hi,

Okey, thanks i will create one while for each case.
Thanks!

Alejandro13C, 2024-06-09:
Not an issue.
Use proper structure.

closed 2024-06-09 by Alejandro13C