olikraus/u8g2 #2353: OLED flicker

patrickwasp, 2024-01-15:
Hi,

I'm encountering a strange issue with my Adafruit 128x64 monochrome OLED screen. After writing to it once during the setup, I've noticed it flickers, even though I'm not continuously updating the content. Has anyone experienced something similar or knows what might be causing this?

It looks solid when looking at it with my eyes but it still causes discomfort. It's subtle flickering, something like I've seen with low-frequency PWM in the past.

https://github.com/olikraus/u8glib/assets/70671760/53478162-81d2-4a9c-a054-62886fe50371

```
#include <U8g2lib.h>
U8G2_SH1107_64X128_F_HW_I2C u8g2(U8G2_R1);

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_mr);

  u8g2.clearBuffer();                  // clear the internal memory
  u8g2.setFont(u8g2_font_6x10_mr);  // choose a suitable font
  u8g2.drawStr(0, 10, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 20, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 30, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 40, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 50, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 60, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.drawStr(0, 70, "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  u8g2.sendBuffer();  // transfer internal memory to the display
}

void loop(void) {}
```

A similar sketch using Adafruit libraries refreshes the screen faster, fast enough for me not to be bothered by this.

```
#include <Adafruit_GFX.h>
#include <Adafruit_SH110X.h>

voide setup() {
  display.begin(0x3C, true);
  display.display();
  display.clearDisplay();
  display.display();
  display.setRotation(1);
  display.setTextSize(1);
  display.setTextColor(SH110X_WHITE);
  display.setCursor(0,0);
  display.print("0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  display.display(); 
}

void loop(void){}
```

olikraus, 2024-01-19:
Each display has a different setup. The u8g2 originally was written for a different display. The above mentioned Adafruit display is described here: https://cdn-shop.adafruit.com/product-files/4650/4650_C14586.pdf
On page 17, the value for the related command 0xd5 is 0x41, however u8g2 will use 0x51 instead:
 https://github.com/olikraus/u8g2/blob/c4f9cd9f8717661c46be16bfbcb0017d785db3c1/csrc/u8x8_d_sh1107.c#L258

You could try to send the modified value with the sendF command:
```
u8g2.sendF("ca", 0xd5, 0x41);
```

See the u8g2 sendF command and the sh1107 controller datasheet for more details, but maybe 0x31 instead of 0x41 may improve this even more.

closed 2024-03-21 by patrickwasp

VayuDev, 2024-10-19:
Just for future people finding this: I found this through issue google and tried a lot of different values, `u8g2.sendF("ca", 0xd5, 0xF0);` produced the best result by far. No flickering visible anymore to me. Thanks at @olikraus!