olikraus/u8g2 #2360: Arabic font in 7 or 8px height

pgeiem, 2024-01-26:
I am looking for a 7 or 8px height font that include arabic letters (unicode arabic presentation forms-B, FE70 to FEFF).
Do you have any or any links where I can found one ? 
The smallest I have found until now is 10px height

Thanks in advance

olikraus, 2024-01-26:
I also checked the ".bdf" files, but smallest size is 12 (cu12.bdf).
Except for what is installed here in u8g2, I don't know any other links. 
In cause you found a suitable bdf font, we can add the same to u8g2.

pgeiem, 2024-01-26:
Ok thanks for your feedback, it's perhaps impossible to made a readable font so small for arabic.
Would be nice to have a feedback from a native arabic.