olikraus/u8g2 #2363: Support for MD240128A6W-FPTLW (ST7586S 240x128)

TheRealJaboo, 2024-01-31:
Datasheet [here](https://uk.farnell.com/midas/md240128a6w-fptlw/graphic-lcd-cob-fstn-240-x-128p/dp/3565074#:~:text=MD240128A6W%2DFPTLW%20%C2%A0%20Datasheet). Display uses parallel 6800 and has resolution 240x128. As it uses display driver ST7586S I got good results with the YMC240160 constructor once i changed some info. It works well when setFlipMode is 1, but not 0. There looks to be a y offset and I don't know where to edit this.

In the YMC240160 constructor I changed some information in`static const u8x8_display_info_t u8x8_st7586s_ymc240160_display_info =` which starts on line 133. I changed the tile height (not sure if this was necessary) to 16 and the pixel height to 128.
I also changed every instance of 0x09F (referring to the old height of 160-1) to 0x07F (128-1) in the section `static const uint8_t u8x8_d_st7586s_ymc240160_init_seq[] = {` starting line 74.

The results of this are below using the "FlipMode" example:
[image]
[image]

Any help on properly integrating this display would be greatly appreciated. Please let me know if any essential data is missing.
[modified YMC240160 constructor in txt format](https://github.com/olikraus/u8g2/files/14112923/u8x8_d_st7586s_ymc240160.txt)

olikraus, 2024-02-01:
For the offset in flip mode 0, you probably need to change the start line in c-code line 61:

https://github.com/olikraus/u8g2/blob/8430f65c9e2bc7ca393bf0e9d8cf58a84ae081d7/csrc/u8x8_d_st7586s_ymc240160.c#L56-L74

If you have all setup done and if all works as expected, you could post the structures here so that I can create a new constructor.

TheRealJaboo, 2024-02-06:
Yes, looks like the start line was the issue, and I changed the flipmode 0 and 1 commands so 0 is the default rotation. I have attached the new constructor.
Just to double check, should R0 F1 be flipped in the x AND y axis like in the image below? I was struggling to get just a flip in the x axis without a strange offset.
[image]
[image]
[MODIFIED_u8x8_d_st7586s_ymc240160.txt](https://github.com/olikraus/u8g2/files/14181169/MODIFIED_u8x8_d_st7586s_ymc240160.txt)

olikraus, 2024-02-06:
The second image seems to be mirrored by the hardware. Probably the COM/SEG direction is wrong, so in line 69 some other value is required.
Maybe try 0xc8 instead of 0x80 for the Scan Direction Setting.

TheRealJaboo, 2024-02-07:
Hi again Oli, thank you for the help. Unfortunately after trying this I still get a strange output with some sort of x-offset that I can't work out how to fix.
[image]
[image]
Looking through the datasheet it looks like you can only change the two most significant bits MX and MY, but strangely using 0x0c0 instead of 0x0c8 actually give slightly different results with a different offset for the first and last lines compared to the middle. Unfortunately I don't have an image for this.
Any ideas how I can either give the display an x offset to realign the image i included?

TheRealJaboo, 2024-02-13:
Okay after a lot more trial and error I have found how to fix it, but my programming knowledge is too limited to figure out how to actually do it. 
The scan direction should be 0x040 for flip1, but this causes some issues with offset that I mentioned before. These issues were fixed by manually changing the column addresses (see new code attached). For flip0 it uses the first 80 columns (as 80*3=240px) but for flip1 i have changed it to use the **LAST** 80 columns to get over the offset. Not sure if this is a good way of doing it, but it does work mostly as intended.
The only problem with this is that when the display is flipped, every three pixels are written the wrong way as shown below.
[image]
[image]
The below two images are with 0x040 for the scan direction but with the sets of 3 bits reversed.
[image]
[image]

The lines I changed to achieve this were 198 and 292 in the modified constructor attached. I changed the value 0xC0 in line 198 to 0x3 and on line 202 I changed 0x3 to 0xC0.

All that needs done now is to find a way to change between these two methods depending on the flipmode used. I'm not sure how to do this myself, but I believe all the work that requires testing on the display is complete so if you're able to find a way to do that it should work exactly as intended.

Please let me know if there's anything else at all I can help with, I really appreciate your work with this library!

olikraus, 2024-02-13:
> but my programming knowledge is too limited to figure out how to actually do it.

I think it is more kind of guessing what is meant and intended by those incomplete datasheets.

Thanks for the code update, however, it looks like that the last pic is still mirrored. 

Is the current version good enough for your purpose?
Shell I create a new constructor for your display... still knowing that one of the modes is broken?

TheRealJaboo, 2024-02-13:
I'm not too sure I understand, if it's using flipmode 1 shouldn't it be mirrored across the X axis like that like if you looked at it through a mirror?
But yes, I would appreciate a constructor being added even with a broken mode as I only need the default rotation and flip for my purpose.

olikraus, 2024-02-13:
> I'm not too sure I understand, if it's using flipmode 1 shouldn't it be mirrored across the X axis like that like if you looked at it through a mirror?

In your last picture the text seems to be mirrored...

> But yes, I would appreciate a constructor

ok, then let's do so...

TheRealJaboo, 2024-02-14:
> In your last picture the text seems to be mirrored...

Could you please explain exactly what changing flipmode *should* do because I can find [this example](https://www.youtube.com/watch?v=gfj2lF2N0Og) on youtube which rotates the image 180 degrees with text still legibly "forward" but I was under the impression it just mirrored the entire display across the x axis like if you were to hold a mirror up to it resulting in "backwards" text
If it's the former and only requires 180 degree rotation with text still forwards this is very easy for me to do and I can upload a new constructor soon with this behaviour.

olikraus, 2024-02-14:
Exactly, flip 0/1 should just rotate the complete image by 180 degree, no mirror should be there. It is similar to U8G2_R0 vs U8G2_R2, however the idea if the flip mode is to support this by reprogramming the internal hardware registers of the display (U8G2_R2 is a pure software solution).
The hardware solution with the two flip mode would also then work with U8X8.

olikraus, 2024-08-31:
new constructor: U8G2_ST7586S_MD240128_1_4W_HW_SPI

closed 2024-08-31 by olikraus