olikraus/u8g2 #2364: Support for ST7567S

paolovr970, 2024-02-03:
Hi,
I bought this display, is i2c. 

https://a.aliexpress.com/_EuXvhAd

Can you help for construct? in the exampe helloworld for HW i2C there are only these 2:
//U8G2_ST7567_64X32_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); 
//U8G2_ST7567_HEM6432_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); 

thanks.
p.

paolovr970, 2024-02-03:
[image]

olikraus, 2024-02-04:
Actually each constructor has also a I2C version if I2C is supported by the controller. The constructor list in the examples are not complete and the full list is here (as stated in each example): https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

I suggest to test with 
 * U8G2_ST7567_JLX12864_1_SW_I2C
 * U8G2_ST7567_ENH_DG128064_1_SW_I2C
 * U8G2_ST7567_ENH_DG128064I_1_SW_I2C
 * U8G2_ST7567_OS12864_1_SW_I2C

You can also later switch to the HW version of I2C, but getting your display work with the SW_I2C constructor seems to usually easier.

paolovr970, 2024-02-05:
Hi,
thank you very much the THIRD CONSTRUCT is PERFECT!
I attach the code used FOR SCROLLING A TEXT (out of the main loop for a message that can change randomly).
The CODE works (there a variable to delay the scroll), but it flashes a bit.
Have you any suggestion on how to improve it? 
thanks. 
p.

olikraus, 2024-02-05:
> Have you any suggestion on how to improve it?

Difficult without seeing the code.

A working example for scrolling is here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Shennong/Shennong.ino

paolovr970, 2024-02-06:
Here the CODE:

`#include <U8g2lib.h>
#include <Wire.h>

u8g2_uint_t offset;     // current offset for the scrolling text
u8g2_uint_t width;      // pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
u8g2_uint_t x;
char lcdScrolltext [26] = "SETUP:please wait "; uint8_t templcd = 0;

U8G2_ST7567_ENH_DG128064I_1_HW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);  // LCD 7567S

/*=== SETUP ================================*/
void setup() {
  Serial.begin(115200); delay(3000); // open serial monitor & wait for connection
  Serial.println(F("\r\n\r\n *** LCD U8G2 example scrolling ***"));
  //Wire.begin(); // initialise I2C
  lcd_init();
} /*END SET UP */

/*=== MAIN LOOP ================================*/
void loop() {
  lcd_display();
  delay(20);
} /*END LOOP */

/*=== lcd_init ================================*/
void lcd_init() {
  u8g2.setI2CAddress(0x3F * 2); // LCD
  //u8g2.setBusClock(100000); // speed
  u8g2.begin();  
  u8g2.enableUTF8Print();
  u8g2.setFontMode(0);             // enable transparent mode, which is faster
  //u8g2.setContrast (60);           // set contrast
  u8g2.clearBuffer();              // clear the internal memory
  Serial.println(F("U8G2: setup complete OK."));
} /*END lcd_init */

/*=== lcd_display ================================*/
void lcd_display() {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_6x10_tr);
    u8g2.setCursor(0, 7); u8g2.print("Hello World");

    // code for last line scroling
    x = offset; 

    u8g2.setFont(u8g2_font_5x7_mr);   // set the target font
    width = u8g2.getUTF8Width(lcdScrolltext);		// calculate the pixel width of the text
    do {                // repeated drawing of the scrolling text...
      u8g2.drawUTF8(x, 63, lcdScrolltext);     // draw the scolling text
      x += width;           // add the pixel width of the scrolling text
    } while( x < u8g2.getDisplayWidth() );    // draw again until the complete display is filled

  } while ( u8g2.nextPage() );

  templcd ++;  // scroll by one pixel depending on templcd
  if (templcd > 3) {offset-=1; templcd = 0;}
  
  if ( (u8g2_uint_t)offset < (u8g2_uint_t)-width ) offset = 0; // start over again
} /*END lcd_display*/

`

olikraus, 2024-02-06:
It mostly looks ok to me. Not sure what kind of flashing you mean... but maybe some remarks:

- `U8G2_ST7567_ENH_DG128064I_1_HW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);` somehow the arg's don't fit. The HW I2C constructor does not expect SCL and SDA arguments
- Maybe to reduce the flashing, you could improve speed. there are two options: 
   * use `U8G2_ST7567_ENH_DG128064I_2_HW_I2C` or `U8G2_ST7567_ENH_DG128064I_F_HW_I2C` 
   * precalculate "width" outside of the outer loop. 
   * To me templcd just further delays the output by redrawing the same screen content multiple times, so maybe achieve this by extending the delay(20)

kittenarmy, 2024-06-21:
I have an ST7567 variant I2C display. I tried using these 4 as per earlier post but none of them work with the Hello World example. 
-     U8G2_ST7567_JLX12864_1_SW_I2C
-     U8G2_ST7567_ENH_DG128064_1_SW_I2C
-     U8G2_ST7567_ENH_DG128064I_1_SW_I2C
-     U8G2_ST7567_OS12864_1_SW_I2C

Just to make sure, how should I connect it to Arduino Nano with this library? I have the display connected to I2C pins A4, A5 and putting `U8G2_ST7567_OS12864_1_SW_I2C u8g2(U8G2_R0, 19, 18);`