olikraus/u8g2 #2367: SSD1362 256x64 possible to demonstrate the text message display with scrolling mode?

Yuentat123, 2024-02-07:
I want to do a vertical scrolling which can display any string text to the oled.  But unfortunately, I had tried few way by import u8g2 or wire.h libraries but it will crash and uncompatible. Of course, I also tried other ways without import any library but it still failed. Any suggestion or solution for my situation? Thanks.

Yuentat123, 2024-02-07:
Is anyone done successful before? Please advise.

olikraus, 2024-02-07:
I created an example for scrolling here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Shennong/Shennong.ino

Yuentat123, 2024-02-08:
@olikraus Thanks for your solution, but i need to download the u8g2 library in this case? But i had tried to download the u8g2 library, import and include it into my project before, but when i try to run, it failed and pop up many error as well. Please advise for this problem, thanks.

Yuentat123, 2024-02-08:
This code should be provided for SSD1306, right? But i currently using SSD1362_V1.0 256x64, and I need to import both of the u8g2 and wire library into my project as well?
/* U8g2 Project: SSD1306 Test Board */
  pinMode(10, OUTPUT);
  pinMode(9, OUTPUT);
  digitalWrite(10, 0);
  digitalWrite(9, 0);

Yuentat123, 2024-02-08:
@olikraus Can you provide me the code sample in .c file? as I'm using C language instead of C++. Another things is I not using the Arduino program software will it affect me to import the u8g2 and Wire library into my project? Thanks for your time.

olikraus, 2024-02-08:
U8g2 is an arduino library. You need to use the Arduino IDE. If you use pure c then you need an replacement for the Wire library, because the Wire lib is specific to Arduino. I strongly suggest to use Arduino IDE.

olikraus, 2024-02-08:
Step by step instructions for C is here: https://github.com/olikraus/u8g2/wiki/u8g2as7

Yuentat123, 2024-02-09:
@olikraus Thanks for your advise and the explanation. I actually want to use others ide to do this feature and the resources is quite limited on the browser so this is the only way I can request for help. By the way, thanks very much.

olikraus, 2024-02-09:
Well, as mentioned U8g2 had been written for Arduino, so there is not much support possible from my end if you use any other IDE.

Yuentat123, 2024-02-15:
@olikraus Ok, btw thanks for your help and explanation.

Yuentat123, 2024-02-15:
And I will close the comment. Thanks again.

closed 2024-02-15 by Yuentat123