olikraus/u8g2 #2368: SSD1312 128x32 issue

Rick-NL, 2024-02-08:
I have aquired a 128x32 size, 1.71" OLED Display to get a better readabilitywithout glasses than the original 0.91" SSD1306 Display.
I don't get the code compiled with the SSD1312_128x64 (or 128x32) options, with the original SSD1306 I get this:
[image]
Is there a solution that I could implement?

Rick-NL, 2024-02-08:
This is the article on Aliexpress
https://www.aliexpress.us/item/3256805486386510.html?
Rick

olikraus, 2024-02-09:
hmm... the controller SSD1312 is not supported at the moment by u8g2

olikraus, 2024-02-09:
I have ordered one of those display. It will be in my lab in arround 5 weeks.

Rick-NL, 2024-02-09:
That would be super nice, thanks I will wait and hope!

Rick-NL, 2024-02-09:
I can send you one from The Netherlands to Gemany, that would be quicker. When you want so please PM your address.

olikraus, 2024-02-09:
Nice offer, but it's ordered already. Sometimes those devices arrive also more earlier.

Rick-NL, 2024-02-09:
OK, thats fine

nikolajovickg, 2024-03-11:
Workaround that works for me until official
support is SSD1312_128x64 constructor with y offset of 16 pixels (so effective display mapping is x: 0-127, y: 16-47)
[image]

wacd, 2024-03-21:
> Nice offer, but it's ordered already. Sometimes those devices arrive also more earlier.

Hello Sir, is it compatible now i have the same issue with nikolajovickg.

olikraus, 2024-03-31:
meanwhile the display arrived in my lab, but I don't know when I will have time to work on this.

Rick-NL, 2024-04-01:
Good it did arrive.
Many thanks for al the work you do!
I will await your reaction,
Rick

olikraus, 2024-04-14:
I have create a new constructor for this display `U8G2_SSD1312_128X32` 
``` C++
U8G2_SSD1312_128X32_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SSD1312_128X32_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ 8);
U8G2_SSD1312_128X32_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);
``` 
I have received a SPI display which works as expected.
Maybe someone can test the I2C interface.
The latest u8g2 beta 2.35.16 includes the update...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Rick-NL, 2024-04-14:
Thanks a lot,
I will check it soon on i2c
Rick

Rick-NL, 2024-04-23:
Hi Olikraus, it took a few days before I had the chance to test the new files.
Being not a very good programmer, it took me a few hours to find out how easy it finally was.
 Needed to use the full buffer option, the result is perfect.
I use it on HW I2C with the constructor : U8G2_SSD1312_128X32_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);
[image]
Thanks a lot

olikraus, 2024-04-28:
Amazing, looks good. Thanks for the feedback!

closed 2024-04-28 by olikraus

VoladorUK, 2024-10-01:
Hi Rick, I think I have the same OLED I2C 128x32 as you running on a MEGA but have not got it working with Oli's new library. I've had mixed success with other libs but garbled results on the screen. There is almost no documentation and I don't know how I should be using reset pin 8 with the module (if that is significant) as it only has the four pins. Could you share your constructor and and example of the coordinates you used to display things on the screen? Thanks in advance

Rick-NL, 2024-10-05:
Hi there,I am now in France as the Dutch team manager for the RC model sailplane World Campionship.Whe I am back or my be even somwwhat earlier I will send you my files that made it work on an Arduino Pro Mini clone.Regards, Rick Ruijsink

VoladorUK, 2024-10-05:
Hi Rick, thank you for your reply, I got it working (by re-reading the command you posted).
Cheers
Chris

lauaall, 2025-04-25:
@Rick-NL can you please provide the code you used. Im having trouble with I2C.

Rick-NL, 2025-04-25:
Here you can download the project.
https://www.dropbox.com/scl/fo/w3aykkvzzqn2ikwo8ikui/AEj0fWOu6WtHqgJ2LLTRyA0?rlkey=o6co63fu3bj2nnq0sv3w02aqt&dl=0

Succes

lauaall, 2025-04-25:
Thanks. i tested the I2C pins and seems like its just a problem with my ESP32CAM pcb. I also tested it with a ESP8266 and still didnt work. Maybe my oled is just faulty. ....

lauaall, 2025-04-26:
yes finally got it working :D if anyone else is looking for this code in the future then here you go. Im using ESP32CAM so you cant assign a random pin for I2C working ones are 15, 14.

```
#include <Wire.h>
#include <U8g2lib.h>

#define SDA_PIN 15
#define SCL_PIN 14
#define RESET_PIN 2 

// U8g2 constructor for SSD1306 using software I2C
U8G2_SSD1312_128X32_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL_PIN, /* data=*/ SDA_PIN, /* reset=*/ RESET_PIN);

void setup() {
  Serial.begin(115200);
  while (!Serial); 
  Serial.println("Starting OLED test");
  pinMode(RESET_PIN, OUTPUT);
  digitalWrite(RESET_PIN, LOW); 
  delay(10);
  digitalWrite(RESET_PIN, HIGH); 
  delay(10);
  Wire.begin(SDA_PIN, SCL_PIN, 100000);  // 100 kHz
  Serial.println("I2C Initialized on SDA: GPIO 15, SCL: GPIO 14");
  u8g2.begin();
  Serial.println("U8g2 initialized");
  Serial.println("Charge pump enabled");
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_profont12_mr);
  u8g2.drawStr(0, 10, "Hello, OLED!");
  u8g2.sendBuffer();
  Serial.println("Test pattern sent to OLED");
}

void loop() {

}
```