olikraus/u8g2 #2370: ST7567 NHD-C12864WO-B1TFH#-M

SVW1986, 2024-02-14:
I could use some help getting the library to work with my display
:
I got the following display for my project: NHD-C12864WO-B1TFH#-M
datasheet: https://newhavendisplay.com/content/specs/NHD-C12864WO-B1TFH-M.pdf

It has an ST7567 controller.

The display is connected through the SPI interface with D/C.
One particular item: I ran out of pins so the RESET line is connected to a port extender. The reset line goes high right before the u8g2.begin() statement.
[image]

I've tried all of the _1 and _F constructors that mention the controller. I did get some fixed pixels with some constructors and vertical lines with some lines flickering on others. the behavior when looking at the _1 and _F constructors was the same (I did update the programming to account for the different contructors)

I've also tried to lower the clock frequency: I've tried 8MHz, 4MHz and 2MHz.

The controller I'm using is the ESP32-WROOM-DA module and I'm using the Arduino IDE to program it. (so the library was also downloaded and installed through the arduino IDE.)

Any Ideas on what to try would be greatly appreciated!

Thanks

olikraus, 2024-02-14:
maybe you can also share some code, which you have used along with a picture on how it looks at the moment.

SVW1986, 2024-02-14:
around 2 minutes after I posted this (because that is usually when you find it right 👍) I found a small error in the code. (a copy paste error with the pin definition of D/C)

I will re-test everything first before I bother you or anyone with questions 😮‍💨 

Thanks for the quick reply though 🥇

SVW1986, 2024-02-14:
I got it working with the following constructor:
U8G2_ST7567_OS12864_F_4W_HW_SPI

It is now showing hello world! nicely on the display

The conclusion:
* Don't code when you're tired
* Double check the pin definitions
* I'm stupid 🤪

olikraus, 2024-02-14:
BTW: C5 should be close to pins 14 and 15