olikraus/u8g2 #2372: Some issues in my uc1601s 128X64  via I2C

hongshimizi, 2024-02-14:
Hi

I'm trying to use u8g2 and UC1601s 128x64 display on my stm32L496 but it still doesn't work well, does anyone have any experience with UC1601s 128x64 Display ?.
[image]
And the datasheet below
[UC1601s_A1.5.pdf](https://github.com/olikraus/u8g2/files/14283961/UC1601s_A1.5.pdf)
my setting below
static const uint8_t u8x8_d_uc1601_128x64_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0xe9),            			/* LCD Bias: 0xe8: 6, 0xe9: 7, 0xea: 8, 0xeb: 9 */
  //U8X8_C(0x023),            			/* 0x020...0x023 only for UC1601, not for UC1601s */
	

  //U8X8_C(0x02e),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */
  U8X8_C(0x024),		                /* Temperature Compenstation, default: 0x24 */
  U8X8_C(0x089),		                /* RAM address ctrl, default: 0x89 */
  U8X8_C(0x0c4),		                /* RAM mapping ctrl */
  U8X8_C(0x0a0),		                /* Frame Rate, 0x0a0 or 0x0a1 */
  U8X8_CA(0x081, 0x0df),		/* set contrast */
  U8X8_C(0x02e),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */	
  U8X8_C(0x040),		                /* set display start: See Issue 1501 */
  
  U8X8_C(0x0a6),		                /* normal display */
   
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */

I just do this cmd to work as a "Hello world" ...

       u8g2_SendBuffer(&u8g2);
       u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
       u8g2_DrawStr(&u8g2, 2, 60, "Hello World!");
       u8g2_SendBuffer(&u8g2);

But not work well, does anyone help me ?

olikraus, 2024-02-14:
Did you try to lower the contrast? (u8g2_SetContrast()) In your image it looks like, that the contrast level is too high. This would make any content invisible.

The other issue might be, that your low level driver for u8g2 is not working as expected.
U8g2 might work more reliable with STM32Duino (https://github.com/stm32duino/Arduino_Core_STM32).

hongshimizi, 2024-02-15:
I've made changes here U8X8_CA(0x081, 0x006),//0x0df),		/* set contrast */  in“  u8x8_d_uc1601.c”
But the effect is the same

hongshimizi, 2024-02-15:
Is there a c codebase in STM32Duino?  I need to work in Keil environment

hongshimizi, 2024-02-15:
Luckily， I find this code may work well. 
void initialize(void)
{
	reset();

	writecommand(0xE2);//System Reset 
	flash(20);
	writecommand(0x24);//Set Temperature Compensation  -0.05%/C
	writecommand(0xC4);//Set LCD Mapping Control  MX = 0 MY = 0
	writecommand(0xA0);//Set Frame Rate 76 fps
	writecommand(0xE9);//Set LCD Bias Ratio 01b= 1/7
	writecommand(0x81);//Set VBIAS Potentiometer
	writecommand(vop);

//	writecommand(0x29);//Set Panel Loading  01b > 15nF 
    writecommand(0xf1);
	writecommand(0x1f);

	writecommand(0xf2);
	writecommand(0x00);

	writecommand(0xf1);
	writecommand(0x1f);

	writecommand(0x2F);//Set Pump Control  11b= Internal VLCD ( standard ) 
	writecommand(0xAF);//Set Display Enable
}

void display(uchar dat1, uchar dat2)
{
	uchar i,j;
	for(i = 0; i < PAGE; i++)
	{
		writecommand(0xB0 + i);//set page address
		writecommand(0x00);//Set Column Address LSB CA[3:0] 
		writecommand(0x10);//Set Column Address MSB CA[7:4] 

		for(j = 0; j < (SEG >> 1); j++)
		{
			writedata(dat1);
			writedata(dat2);
		}
	}
}

void display_pic(uchar* pic)
{
	uchar i,j;
	for(i = 0; i < PAGE; i++)
	{
		writecommand(0xB0 + i);//set page address
		writecommand(0x00);//Set Column Address LSB CA[3:0] 
		writecommand(0x10);//Set Column Address MSB CA[7:4] 

		for(j = 0; j < SEG; j++)
		{
			writedata(*pic++);
		}
	}
}

void main(void)
{
	flash(2000);
	dir=1 ;
	coe=0 ;
	doe=1 ;
	koe=1 ;
	sent(VDD,IDDmax,VOPmin,VOPmax,project);
	flash(2000);

	vop = 152;
	initialize();
	
	display(0xFF,0xFF);
	delay1(0xFFFF);
	
	vop = 168;//154=10 V
	initialize();

	while(1)
	{
		display(0xFF,0xFF);
		delay1(50000);delay1(50000);
		
		display(0x00,0x00);
		delay1(50000);delay1(50000);
		
		display(0xAA,0x55);
		delay1(50000);delay1(50000);
		
		display(0x55,0xAA);
		delay1(50000);delay1(50000);
		
		display_pic(P111);
		delay1(50000);delay1(50000);
		
		display_pic(PQQQ);
		delay1(50000);delay1(50000);
	}
}
This configuration is correct，But there are few comments, I can't understand it，Is it significantly different from my configuration that would cause this display result？

olikraus, 2024-02-15:
Looks like there is no data transfer. How does your u8g2 setup looks like? Especially I assume we need to discuss the gpio and delay procedure (and maybe the byte transfer callback) more in detail.

Again: We are leaving the standard u8g2 configuration towards a custom HAL environment: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

hongshimizi, 2024-02-15:
You mean these?
void u8g2_Setup_uc1601_i2c_128x64_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_uc1601_128x64, u8x8_cad_uc16xx_i2c, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_16_8_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}

uint8_t u8x8_byte_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    /* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
    static uint8_t buffer[32];
    static uint8_t buf_idx;
    uint8_t *data;

    switch (msg)
    {
    case U8X8_MSG_BYTE_INIT: 
        /* add your custom code to init i2c subsystem */
        MX_I2C1_Init(); //I2C???
        break;
    case U8X8_MSG_BYTE_START_TRANSFER:
        buf_idx = 0;
        break;
    case U8X8_MSG_BYTE_SEND:
        data = (uint8_t *)arg_ptr;
        while (arg_int > 0){
            buffer[buf_idx++] = *data;
            data++;
            arg_int--;
        }
        break;
    case U8X8_MSG_BYTE_END_TRANSFER:
        if (HAL_I2C_Master_Transmit(&hi2c1, (OLED_ADDRESS), buffer, buf_idx, 1000) != HAL_OK){
            return 0;
        }
        break;
    case U8X8_MSG_BYTE_SET_DC:
        break;
    default:
        return 0;
    }

    return 1;
}

void delay_us(uint32_t time)
{
    uint32_t i = 8 * time;
    while (i--)
        ;
}

uint8_t u8x8_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    switch (msg)
    {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
        break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
        break;    
    case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
        __NOP();
        break;
    case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
        for (uint16_t n = 0; n < 320; n++)
        {
            __NOP();
        }
        break;
    case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
        HAL_Delay(1);
        break;
    case U8X8_MSG_DELAY_I2C: // arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
        delay_us(5);
        break;                    // arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_I2C_CLOCK: // arg_int=0: Output low at I2C clock pin
        break;                    // arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:  // arg_int=0: Output low at I2C data pin
        break;                    // arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
        u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_NEXT:
        u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_PREV:
        u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_HOME:
        u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
        break;
    default:
        u8x8_SetGPIOResult(u8x8, 1); // default return value
        break;
    }
    return 1;
}
void u8g2Init(u8g2_t *u8g2)
{
	u8g2_Setup_uc1601_i2c_128x64_f(u8g2, U8G2_R0, u8x8_byte_hw_i2c, u8x8_gpio_and_delay); // ??? u8g2 ???
	u8g2_InitDisplay(u8g2);                                                                       // ??????????????,??????,?????????
	u8g2_SetPowerSave(u8g2, 0);                                                                   // ?????
	//u8g2_ClearBuffer(u8g2);
}

I'm not sure what the device address is, but trying the other doesn't even have a screen, it's empty

hongshimizi, 2024-02-16:
Hmm, I just feel like its I2C communication should be useful, after all, I can modify its contrast, and the strange thing is that I change the font and the things that are displayed on the screen are not the same, is it possible that I have a bit of a problem with my GPIO settings?
I use  u8g2_SetFont(&u8g2, u8g2_font_unifont_t_symbols);
[image]
And my GPIO setting
void MX_I2C1_Init(void)
{

  /* USER CODE BEGIN I2C1_Init 0 */

  /* USER CODE END I2C1_Init 0 */

  /* USER CODE BEGIN I2C1_Init 1 */

  /* USER CODE END I2C1_Init 1 */
  hi2c1.Instance = I2C1;
  hi2c1.Init.Timing = 0x10909CEC;
  hi2c1.Init.OwnAddress1 = 0;
  hi2c1.Init.AddressingMode = I2C_ADDRESSINGMODE_7BIT;
  hi2c1.Init.DualAddressMode = I2C_DUALADDRESS_DISABLE;
  hi2c1.Init.OwnAddress2 = 0;
  hi2c1.Init.OwnAddress2Masks = I2C_OA2_NOMASK;
  hi2c1.Init.GeneralCallMode = I2C_GENERALCALL_DISABLE;
  hi2c1.Init.NoStretchMode = I2C_NOSTRETCH_DISABLE;
  if (HAL_I2C_Init(&hi2c1) != HAL_OK)
  {
    Error_Handler();
  }

  /** Configure Analogue filter
  */
  if (HAL_I2CEx_ConfigAnalogFilter(&hi2c1, I2C_ANALOGFILTER_ENABLE) != HAL_OK)
  {
    Error_Handler();
  }

  /** Configure Digital filter
  */
  if (HAL_I2CEx_ConfigDigitalFilter(&hi2c1, 0) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN I2C1_Init 2 */

  /* USER CODE END I2C1_Init 2 */

}

void HAL_I2C_MspInit(I2C_HandleTypeDef* i2cHandle)
{

  GPIO_InitTypeDef GPIO_InitStruct = {0};
  RCC_PeriphCLKInitTypeDef PeriphClkInit = {0};
  if(i2cHandle->Instance==I2C1)
  {
  /* USER CODE BEGIN I2C1_MspInit 0 */

  /* USER CODE END I2C1_MspInit 0 */

  /** Initializes the peripherals clock
  */
    PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_I2C1;
    PeriphClkInit.I2c1ClockSelection = RCC_I2C1CLKSOURCE_SYSCLK;
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK)
    {
      Error_Handler();
    }

    __HAL_RCC_GPIOB_CLK_ENABLE();
    /**I2C1 GPIO Configuration
    PB6     ------> I2C1_SCL
    PB7     ------> I2C1_SDA
    */
    GPIO_InitStruct.Pin = GPIO_PIN_6|GPIO_PIN_7;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_OD;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF4_I2C1;
    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

    /* I2C1 clock enable */
    __HAL_RCC_I2C1_CLK_ENABLE();

    /* I2C1 interrupt Init */
    HAL_NVIC_SetPriority(I2C1_EV_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(I2C1_EV_IRQn);
  /* USER CODE BEGIN I2C1_MspInit 1 */

  /* USER CODE END I2C1_MspInit 1 */
  }
}

hongshimizi, 2024-02-16:
I'm thinking about whether I should use GPIO for virtual I2C, is there a more complete example for me to refer to?

olikraus, 2024-02-16:
For a software based "virtual I2C" you just need to provide the gpio and delay procedure: see my previous comment above or here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

As a byte procedure you can use the predefined byte level procedure "u8x8_byte_sw_i2c" : https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c

hongshimizi, 2024-02-22:
Thank you very much for your reply！After I switched to software i2c, there was something on the screen. Maybe there's something wrong with my setting of hardware i2c .But it seems that there is still a bit of a problem.,The thing shown is reversed.,And there's a goof.
[image]

olikraus, 2024-02-26:
What exactly is the question?

hongshimizi, 2024-02-29:
[image]
I compared the difference between hardware I2C and software I2C when sending data, I only sent 0x70, 0xaa, 0xbb. I see that there is a gap between each byte of software I2C, I want to ask how hardware I2C gets this interval?

olikraus, 2024-02-29:
How can I tell you how your hardware i2c procedure works? This is beyond my skills.

closed 2024-05-01 by hongshimizi