olikraus/u8g2 #2375: Update: font_tiny5

Gissio, 2024-02-19:
Hallo Oli:

Some weeks ago I updated font_tiny5. You can find the latest release here:

https://github.com/Gissio/font_tiny5

Gissio, 2024-02-19:
It is already part of u8g2! https://github.com/olikraus/u8g2/wiki/fntgrpu8g#tiny5

Could you also add some reference to the font's repository? https://github.com/Gissio/font_tiny5

olikraus, 2024-02-20:
Oh, i remember, it is a followup of #2185

yes, sure. I just need to see how to add the reference to your repo. Thanks a lot!

Gissio, 2024-04-18:
I've just wanted to let you know that I've added greek and cyrillic to tiny5!

olikraus, 2024-04-27:
Thanks!
I think i didn't do anything regarding this topic, but it's on the list for the next release.

Gissio, 2024-04-27:
BTW, this is a bit offtopic but might interest you. Some time ago I wrote a rendering library for color LCD displays (also monochrome) with anti-aliased font rendering and lightweight font compression. I'm using many ideas of u8g2, and added a second-level RLE. In some cases, it even outperforms u8g2's compression!

https://github.com/Gissio/mcu-renderer

olikraus, 2024-04-28:
wow, nice, project "starred". Nice comparison tables and example pictures.

Gissio, 2024-06-16:
I don’t mean to be annoying, but Tiny5 made it to Google Fonts!

https://fonts.google.com/specimen/Tiny5

olikraus, 2024-06-16:
nice :+1:

olikraus, 2024-09-01:
amazing font work...

olikraus, 2024-09-01:
https://github.com/olikraus/u8g2/wiki/fntgrpu8g#tiny5

closed 2024-09-01 by olikraus

Gissio, 2024-09-01:
Hi Oliver:

Danke schön fürs Kompliment!

One more thing: u8g2_font_tiny5_tf needs to be updated as well, as I corrected several characters in Unicode range 0xa1-0xff.

Gissio, 2024-09-01:
Hi Oliver:

Danke fürs Kompliment!

One more thing: u8g2_font_tiny5_tf and u8g2_font_tiny5_tr need to be updated as well, as I corrected several characters in Unicode range 0x20-0xff.

olikraus, 2024-09-01:
> Danke fürs Kompliment!

:-)

> One more thing: u8g2_font_tiny5_tf and u8g2_font_tiny5_tr need to be updated as well, as I corrected several characters in Unicode range 0x20-0xff.

I have rebuild all fonts, so also the pictures in the wiki should reflect your changes.

Gissio, 2024-09-01:
I responded as the pictures in the wiki remained unchanged. So, it seems most likely that it's simply a matter of waiting for GitHub to update the URLs!

olikraus, 2024-09-01:
could be the case...

olikraus, 2024-09-08:
Thanks for crosschecking.
Is everything fixed now?

Gissio, 2024-09-08:
Looking great now!