olikraus/u8g2 #2376: font (u8g2_font_t0_11b_tr) issue

mrinal-crypto, 2024-02-20:
[image]

Hi, I'm using "u8g2_font_t0_11b_tr" font, but there is missing one pixel in zero(0) char at (3,0) position like provided image. please fix it if possible. Thank you

olikraus, 2024-02-20:
well... the fonts are usually taken from some external sources.

In this case from this page here: https://people.mpi-inf.mpg.de/~uwe/misc/uw-ttyp0/sample-6x12.html

And as you can see, the "0" is kind of intended like this.  Usually do not update fonts, instead the right way would be to ask the font author to do so. Then I can include the new version of the font.

olikraus, 2024-02-20:
I think you refered this font here: https://people.mpi-inf.mpg.de/~uwe/misc/uw-ttyp0/sample-6x12b.html

So the "missing" pixel is also missing in the original font.

uwsb, 2024-02-21:
When I designed uw-ttyp0-6x12b, I needed some way to distinguish digit 0 from capital O. None of the possibilities that I tried looked really nice, and omitting one pixel at the top turned out to be the most unobtrusive one. So it's intentional.

olikraus, 2024-02-21:
Thanks for clarifying