olikraus/u8g2 #2377: Support for SED1530 (2x) and 33x200 px Display

karlo922, 2024-02-23:
Hi,

I have an old radio (Mercedes AUDIO 30 APS) using 2 SED1530 (33x100) controllers to control a 33x200 display with serial interface (CS, A0, CLK, SI)
Datasheet can be found here: [https://download.epson-europe.com/pub/electronics-de/assp/sed1500/tm_1500.pdf](url)

What do you need from me to help?

Best regards

olikraus, 2024-02-24:
It looks like this is indeed similar to the S1D15300 constructor, so maybe try this constructor:

U8G2_S1D15300_LM6023_1_4W_SW_SPI(U8G2_R0, clock, data, cs, dc, reset) 

This constructor is actually for a 128x64 display, nevertheless you should be able to "see" something on the display. Maybe some contrast adjustment might be required (or use the ContrastTest.ino for testing)

Please post a picture of the result (HelloWorld.ino) so that we can create a new constructor for your display.

karlo922, 2024-02-24:
Thanks - I will do this as soon I have received the contacts to get it connected.

karlo922, 2024-03-06:
Hi!
The display remains black all time - I guess this is due to the voltage regulator circuit selection that may not fit to my display. It does not do any "buzzing" noises which it does when used in the original way of the radio at startup.

I have recorded a startup sequence of the display of the original radio - I hope this helps fixing it.

  | A0 | D7 | D6 | D5 | D4 | D3 | D2 | D1 | D0 |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Reset | 0 | 1 | 1 | 1 | 0 | 0 | 0 | 1 | 0 | 0xE2
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Reset | 0 | 1 | 1 | 1 | 0 | 0 | 0 | 1 | 0 | 0xE2
Pause 1244 µs | 0 | - | - | - | - | - | - | - | - | -
Page Adr. 0 | 0 | 1 | 0 | 1 | 1 | 0 | 0 | 0 | 0 | 0xB0
Set Col Ad hi | 0 | 0 | 0 | 0 | 1 | 0 | 0 | 0 | 0 | 0x10
Set Col Ad lo | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0x00
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Disp Data | 1 | x | x | x | x | x | x | x | x | -
long Pause | - | - | - | - | - | - | - | - | - | -
Disp ON | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 | 1 | 0xAF
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Disp ON | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 | 1 | 0xAF
Pause 90µs | 0 | - | - | - | - | - | - | - | - | -
ADC Select "0" | 0 | 1 | 0 | 1 | 0 | 0 | 0 | 0 | 0 | 0xA0
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
ADC Select "0" | 0 | 1 | 0 | 1 | 0 | 0 | 0 | 0 | 0 | 0xA0
Pause 64µs | 0 | - | - | - | - | - | - | - | - | -
Normal Reverse Display "normal" | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | 0 | 0xA6
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Normal Reverse Display "normal" | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | 0 | 0xA6
Pause 64µs | 0 | - | - | - | - | - | - | - | - | -
Set output Status register "1" | 0 | 1 | 1 | 0 | 0 | 1 | 0 | 0 | 0 | 0xC8
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Set output Status register "1" | 0 | 1 | 1 | 0 | 0 | 1 | 0 | 0 | 0 | 0xC8
Pause 85µs | 0 | - | - | - | - | - | - | - | - | -
Set Electronic control register 10100 | 0 | 1 | 0 | 0 | 1 | 0 | 1 | 0 | 0 | 0x94
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Set Electronic control register 10100 | 0 | 1 | 0 | 0 | 1 | 0 | 1 | 0 | 0 | 0x94
Pause 64µs | 0 | - | - | - | - | - | - | - | - | -
Set Power Control 100 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 | 0 | 0x2C
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Set Power Control 100 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 | 0 | 0x2C
Pause 81µs | 0 | - | - | - | - | - | - | - | - | -
LCD Bias "0" | 0 | 1 | 0 | 1 | 0 | 0 | 0 | 1 | 0 | 0xA2
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
LCD Bias "0" | 0 | 1 | 0 | 1 | 0 | 0 | 0 | 1 | 0 | 0xA2
Pause 311µs | 0 | - | - | - | - | - | - | - | - | -
not documented | 0 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0x80
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
not documented | 0 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0x80
Pause 64µs | 0 | - | - | - | - | - | - | - | - | -
Set Standby "0" - Off | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 | 0 | 0xAC
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Set Standby "0" - Off | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 | 0 | 0xAC
Pause 64µs | 0 | - | - | - | - | - | - | - | - | -
Entire Display ON/OFF - 0 normal | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 | 0 | 0xA4
Pause 20µs | 0 | - | - | - | - | - | - | - | - | -
Entire Display ON/OFF - 0 normal | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 | 0 | 0xA4
End |   |   |   |   |   |   |   |   |   |

olikraus, 2024-03-08:
So also ContrastTest.ino will not at least change the contrast somehow?

karlo922, 2024-03-08:
No - I guess the issue is, that CS needs to be "1" to control the Master. It is now "0" which would only activate the slave chip. How can I change CS to be "1" if active?

karlo922, 2024-03-08:
Found the lines in the code I guess:
```
/* chip_enable_level = */ 1,
  /* chip_disable_level = */ 0,
```
But it did not help. I also now changed the SPI mode to "3" as CLK is active high in the original one.

karlo922, 2024-03-08:
I see a timing difference after the "soft reset" command from the original one. The original one waits 1ms after sending 0xE2. The display does not provide a RES input, so I cannot make use of 
```
/* post_reset_wait_ms = */ 1, 
```

How could I achieve a 1ms waiting after the reset command? I also have the Osci raw data if this helps.

olikraus, 2024-03-08:
Yes, you have to swap the values here: 
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_s1d15300.c#L140-L141

This should be already 1ms: 
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_s1d15300.c#L146

Change it to 2 for 2 ms.

karlo922, 2024-03-08:
If I got your code right, the ```post_reset_wait_ms ``` is only used after resetting via the reset GPIO pins. I do not have any reset input at the display so I disabled them in the constructor. I guess this time is not used after sending the "soft reset - 0xE2" command. But the original display waits after this command for 1ms.

karlo922, 2024-03-10:
Just to see if I get this anyhow working I now just directly send the commands the original unit sends 1:1 to the SPI bus without using your library - it works! So at least all cables, SPI settings and so on are correct. Now I need to see when it stops working if I remove the part the current implementation of the u8g2 cannot handle with this constructor (the master/slave controllers, maybe the waiting time after soft-reset etc.) I will report as soon as I know where the issue lies.

karlo922, 2024-03-11:
I got it working "half" with following constructor:
```
static const u8x8_display_info_t u8x8_s1d15300_lm6023_display_info =
{
  /* chip_enable_level = */ 1,
  /* chip_disable_level = */ 0,
  
  /* post_chip_enable_wait_ns = */ 250,	/*  */
  /* pre_chip_disable_wait_ns = */ 120,	/*  */
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 1, 
  /* sda_setup_time_ns = */ 200,		/* */
  /* sck_pulse_width_ns = */ 200,	/* half of cycle time (100ns according to datasheet), AVR: below 70: 8 MHz, >= 70 --> 4MHz clock */
  /* sck_clock_hz = */ 1000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 3,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 200,	/* st7565 datasheet, table 24, tds8 */
  /* write_pulse_width_ns = */ 200,	/* st7565 datasheet, table 24, tcclw */
  /* tile_width = */ 12,		/* width of 12*8=96 pixel */
  /* tile_height = */ 4,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 4,
  /* pixel_width = */ 96,
  /* pixel_height = */ 32
};
```

And following init procedure:
```
static const uint8_t u8x8_d_s1d15300_lm6023_init_seq[] = {

    U8X8_START_TRANSFER(), /* enable chip, delay is part of the transfer start */

    U8X8_C(0x0e2), /* soft reset */
    U8X8_C(0x0af), /* display on */
    U8X8_C(0x0a0), /* ADC set to normal */
    U8X8_C(0x0a6), /* Display normal */
    U8X8_C(0x0c8), /* Set output Status register "1" */
    U8X8_C(0x094), /* Set Electronic control register 10100 */
    U8X8_C(0x02c), /* Set Power Control 100 */
    U8X8_C(0x0a2), /* CS_1 - LCD Bias "0" */
    U8X8_C(0x080), /* Set Electronic control register 00000 */
    U8X8_C(0x0ac), /* Set Standby "0" - Off */
    U8X8_C(0x0a4), /* all point normal */
    U8X8_C(0x0a6), /* Display normal */
    U8X8_C(0x0b0), /* set page address */
    U8X8_C(0x010), /* set column address upper */
    U8X8_C(0x000), /* set column address lower */

    U8X8_END_TRANSFER(), /* disable chip */
    U8X8_END()           /* end of sequence */  
};
```

**Now there stay 2 questions:**

1. How can I use the full left half display (now 96x32 as I think the tilw_width etc. need to be multiplied by 8, but 100x33 is not dividable by 8?
2. How can I enable the right half of the display which is controlled by the second SED1530 using CS=0?

olikraus, 2024-03-11:
> How can I use the full left half display (now 96x32 as I think the tile_width etc. need to be multiplied by 8, but 100x33 is not dividable by 8?

Just increase the "tile_width" to 13 and assign the correct pixel value 100 into "pixel_width".  

> How can I enable the right half of the display which is controlled by the second SED1530 using CS=0? 

The best way would be to create a second u8g2 object for the second display. Is it correct, that the CS line is inverted for the other display?

karlo922, 2024-03-11:
> The best way would be to create a second u8g2 object for the second display. Is it correct, that the CS line is inverted for the other display?

Yes, CS is inverted for the right half of the display. It's the same display, so it would be great if the code automatically would invert CS when x > 100.

karlo922, 2024-03-11:
Just to be sure we have the same understanding:
The two controllers are mounted in "Master/Slave" configuration on the board behind the display - so you have only 4 wires for driving both controllers simultaneously: DAT, CLK, CS and A0. 

The controllers itself have two CS inputs: !CS1 & CS2
- on the left one (master), !CS1 is connected to GND and CS2 is driven by the CS line which can be set via outside connection
- on the right one (slave),  !CS1 is driven via the CS line from outside and CS2 is connected to VCC

With that configuration, "master controller" will be active if CS = high and "slave controller" will be active if CS = low.

olikraus, 2024-03-14:
Thanks for all the input. However I don't know when I will find time to work on this.

karlo922, 2024-03-14:
Thanks - I maybe try to find out a little bit myself... I guess a staring point could be the one special thing you already have using 3 "displays" ``` u8x8_d_hd44102.c ``` There you use 2 CS lines, I think this is near to what I need with the change that I need SPI and only one CS. But there you already "split" the output after x > threshold.

olikraus, 2024-03-14:
ok... I just decided to look at this a little bit closer.

karlo922, 2024-03-14:
even better! Thanks!

olikraus, 2024-03-14:
As of now I have created two constructors:
U8G2_S1D15300_100X33
U8G2_S1D15300_100X33I

The I version has an inverted CS1 input. 
Can you check whether the two constructors are working as expected?
I also tookover the config from your trace with one exception: I have used 0x02f instead of 0x2c for the power setup.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2024-03-14:
U8g2 has powerfull clipping procedure, so drawing the same content twice on both displays shouldn't be a big delay.
Example taken from here: https://github.com/olikraus/u8g2/issues/1351

``` C++

U8G2_S1D15300_100X33_2_4W_HW_SPI u8g2_0(U8G2_R0, /* cs=*/ 1, /* dc=*/ 2, /* reset=*/ 3);
U8G2_S1D15300_100X33I_2_4W_HW_SPI u8g2_1(U8G2_R0, /* cs=*/ 1, /* dc=*/ 2, /* reset=*/ 3);

U8G2 *u8g2_current;
u8g2_uint_t offset;

void setLeftSide() {
  u8g2_current = &u8g2_0;
  offset = 0;
}
void setRightSide() {
  u8g2_current = &u8g2_1;
  offset = 100;
}

void drawStr(u8g2_uint_t x, u8g2_uint_t y, const char *s) {
  u8g2_current->drawStr(x-offset,y,s);
}

void drawWelcome() {
  u8g2_current->setFont(u8g2_font_ncenB08_tr);
  drawStr(120, 40, "Welcome");
}

void setup(void) {
    Serial.begin(115200);

    u8g2_0.begin();
    u8g2_0.setFlipMode(0);
    u8g2_0.setContrast(255);
    u8g2_0.setFont(u8g2_font_ncenB08_tr);

    u8g2_1.begin();
    u8g2_1.setFlipMode(1);
    u8g2_1.setContrast(255);
    u8g2_1.setFont(u8g2_font_ncenB08_tr);

    Serial.println("Lanuching test...");
}

void loop(void) {

    setLeftSide();
    u8g2_current->firstPage();
    do { drawWelcome(); } 
    while (u8g2_current->nextPage());

    setRightSide();
    u8g2_current->firstPage();
    do { drawWelcome(); } 
    while ( u8g2_current->nextPage());
}
```

karlo922, 2024-03-14:
Can it be that you did not update the u8g2.h? it does not find it

karlo922, 2024-03-14:
U8G2_S1D15300_100X33I_2_4W_HW_SPI:
[image]

If I use the one without "I" nothing is shown.

karlo922, 2024-03-14:
To be more precise:
If the left display part has worked, right after uploading, the right display will show something:
[image]

But after a power-up, only left dsiplay (when using "I") or nothing is active.

olikraus, 2024-03-14:
> Can it be that you did not update the u8g2.h? it does not find it

Usually this is updated automatically

> If the left display part has worked, right after uploading, the right display will show something:

ok... maybe you can also send the code, which should produce this result.

> "Linkes Display oder rechts"

hmm, the display seems to be mirrored, right?

karlo922, 2024-03-14:
This is the code. Once with "I" once without
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
  } while ( u8g2.nextPage() );
  //delay(1000);
}
```

Forget about the .h file comment - my fault. Everything is good.

Yes it is mirrored. It wasn't when I manually changed the constructor myself.

olikraus, 2024-03-14:
> It wasn't when I manually changed the constructor myself.

strange, I didn't change this part.

Nevertheless, to fix the mirror topic, can you do a `u8g2.sendF("c", 0x0a1);` right after `u8g2.begin()`? 
Will this fix the mirrored display?

karlo922, 2024-03-14:
No it does not fix it

olikraus, 2024-03-14:
What about this?
You can change 0x0a0 to 0x0a1 also. It actually should have an effect... hmmm

``` C++
void setup(void) {
  u8g2.begin();  
  u8g2.sendF("c", 0x0a0);
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
  } while ( u8g2.nextPage() );
  //delay(1000);
}
```

karlo922, 2024-03-14:
It mirrors it, but actually it is rotated by 180 degrees

karlo922, 2024-03-14:
a1
[image]

a0
[image]

olikraus, 2024-03-14:
Thanks for the pics. In between i am studiing the datasheet...
I think we have to play around also with the 0x0c0 command.
There are four combinations:
``` C++
u8g2.sendF("cc", 0x0a0, 0x0c0);
u8g2.sendF("cc", 0x0a0, 0x0c8);
u8g2.sendF("cc", 0x0a1, 0x0c0);
u8g2.sendF("cc", 0x0a1, 0x0c8);
```
Which of the four settings will lead to a correct image?

karlo922, 2024-03-14:
This one works:
u8g2.sendF("cc", 0x0a0, 0x0c8);

[image]

olikraus, 2024-03-14:
... then let's have a look at the second display.

``` C++
U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2i.begin();  // start with the I display
  u8g2i.sendF("cc", 0x0a0, 0x0c8);
  u8g2.begin();  
  u8g2.sendF("cc", 0x0a0, 0x0c8);
}

void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
  } while ( u8g2.nextPage() );
}

```
Will this work?

karlo922, 2024-03-14:
Right after upload, some artifacts:
[image]

after power latch:
[image]

olikraus, 2024-03-14:
ok, then we need to see the borders of the display, let's draw a frame:
``` C++
U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2i.begin();  // start with the I display
  u8g2i.sendF("cc", 0x0a0, 0x0c8);
  u8g2.begin();  
  u8g2.sendF("cc", 0x0a0, 0x0c8);
}

void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
    u8g2i.drawFrame(0,0,100,33);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
    u8g2.drawFrame(0,0,100,33);
  } while ( u8g2.nextPage() );
}
```

karlo922, 2024-03-14:
[image]

olikraus, 2024-03-14:
... hmmm more reveres engineering required, than expected...

The goal is to make the frame fully visible. It looks like the frame is moved somehow in x and y, this must be fixed
Moreover I am missing the lower line of the frame. 
Are you sure that this is a 100x33 display, instead it could be 100x32

Lets rotate the text and frame by 180 degree. How will it look?
``` C++
U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2i.begin();  // start with the I display
  u8g2i.sendF("cc", 0x0a1, 0x0c0);
  u8g2.begin();  
  u8g2.sendF("cc", 0x0a1, 0x0c0);
}

void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
    u8g2i.drawFrame(0,0,100,33);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
    u8g2.drawFrame(0,0,100,33);
  } while ( u8g2.nextPage() );
}
```

karlo922, 2024-03-14:
with 
```
void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
    u8g2i.drawFrame(0,0,100,32);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
    u8g2.drawFrame(0,0,98,32);
  } while ( u8g2.nextPage() );
}
```
at the right, 2 columns are on in parallel 
at the left, it does not use the full space
the middle is 2 lines next to each other ->correct I guess

[image]

[image]

karlo922, 2024-03-14:
> ... hmmm more reverse engineering required, than expected...
> 
> The goal is to make the frame fully visible. It looks like the frame is moved somehow in x and y, this must be fixed Moreover I am missing the lower line of the frame. Are you sure that this is a 100x33 display, instead it could be 100x32
> 
> Lets rotate the text and frame by 180 degree. How will it look?
> 
> ```c++
> U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
> U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
> 
> void setup(void) {
>   u8g2i.begin();  // start with the I display
>   u8g2i.sendF("cc", 0x0a1, 0x0c0);
>   u8g2.begin();  
>   u8g2.sendF("cc", 0x0a1, 0x0c0);
> }
> 
> void loop(void) {
>   u8g2i.firstPage();
>   do {
>     u8g2i.setFont(u8g2_font_ncenB10_tr);
>     u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
>     u8g2i.drawFrame(0,0,100,33);
>   } while ( u8g2i.nextPage() );
>   u8g2.firstPage();
>   do {
>     u8g2.setFont(u8g2_font_ncenB10_tr);
>     u8g2.drawStr(0,24,"Linkes Display oder rechts");
>     u8g2.drawFrame(0,0,100,33);
>   } while ( u8g2.nextPage() );
> }
> ```

like that:
[image]

I will look at the manual again, may be x32 display indeed

karlo922, 2024-03-14:
You are right, pin173 is not connected, so we have only 32 columns -> 32 height

olikraus, 2024-03-14:
Yes... the indicator COMS is not supported by u8g2. Anyway it is probably not usable for the graphcis area.
Still the question is, how to get rid of the 2-pixel left gap.

I have created a new release with a new constructor, so 

``` C++
U8G2_S1D15300_100X33_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X33I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2i.begin(); 
  u8g2.begin();  
}

void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"i DISPLAY i DISPLAY");
    u8g2i.drawFrame(0,0,100,32);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Linkes Display oder rechts");
    u8g2.drawFrame(0,0,100,32);
  } while ( u8g2.nextPage() );
}
```
should work, but I don't know how to fix the left gap

latest release:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

karlo922, 2024-03-14:
The new constructor name is not found

karlo922, 2024-03-14:
From the service manual, I also think the left display has 100x32 (COM1..32 connected and SEG0..99) but the right one only has SEG0...98 connected, so this should be a 99x32. 
The we would have a full 199x32 display

olikraus, 2024-03-14:
>  The new constructor name is not found

Seems to be too late...

I created another release...

karlo922, 2024-03-14:
``` lib\U8g2_Arduino-master\src\U8x8lib.h:8798:27: error: 'u8x8_d_s1d15300_100x32i' was not declared in this scope ```
I guess we should work on this another day ;)

karlo922, 2024-03-14:
After fixing the names, I get this. Last for today:
[image]

olikraus, 2024-03-14:
> lib\U8g2_Arduino-master\src\U8x8lib.h:8798:27: error: 'u8x8_d_s1d15300_100x32i' was not declared in this scope

argl, i have created another release...

karlo922, 2024-03-15:
I have tried a little bit around. To see whats happening, I only activated the frame on the "right" display. It's worse than we thought:

```
8G2_S1D15300_100X32_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X32I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2i.begin(); 
  u8g2.begin();  
}

void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"Links");
    //u8g2i.drawFrame(0,0,100,32);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Rechts");
    u8g2.drawFrame(0,0,98,32);
  } while ( u8g2.nextPage() );
}
```

- "97 width" leads to a singlepixel column on the right and 2 inactive columns on the outer right
- "98 width" leads to a double pixel column on the outer right
- "99 width" leads to a double pixel column at display left!
- "100" width leads to show only the horizontal 2 pixels of the frame, but at the left.

99 width:
[image]

The left one works as expected, ```u8g2i.drawFrame(0,0,100,32);``` starts at the third column from left, and shows a clean frame.

**Conclusion:**

- The outer right AND outer left are both connected to the "right" controller, but you cannot control one column, always 2 columns will be "on".
- The right controller really has 99x32 pixel active
- The left controller starts at 3. column from left with 100x32 pixel active

I guess this means we change the "non-i" variant to 97x32 (remove the 2x2 columns at outer left/right which cannot be used single) and just ignore those "missing" 4 columns.

```
void loop(void) {
  u8g2i.firstPage();
  do {
    u8g2i.setFont(u8g2_font_ncenB10_tr);
    u8g2i.drawStr(0,24,"Links");
    u8g2i.drawFrame(0,0,100,32);
  } while ( u8g2i.nextPage() );
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Rechts");
    u8g2.drawFrame(0,0,97,32);
  } while ( u8g2.nextPage() );
}
```

produces:
[image]

olikraus, 2024-03-15:
Looks like a strange HW configuration :-D

I have created another release with additioanl "U8G2_S1D15300_97X32_F_4W_HW_SPI" constructor

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

karlo922, 2024-03-15:
Your new constructor works!
But I do not get the code you showed with clipping to show anything on the display.

karlo922, 2024-03-15:
Got it working:

```
U8G2_S1D15300_97X32_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);
U8G2_S1D15300_100X32I_F_4W_HW_SPI u8g2i(U8G2_R0, /* cs=*/ 9, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);

U8G2 *u8g2_current;
u8g2_uint_t offset;

void setLeftSide() {
  u8g2_current = &u8g2i;
  offset = 0;
}
void setRightSide() {
  u8g2_current = &u8g2;
  offset = 100;
}

void drawStr(u8g2_uint_t x, u8g2_uint_t y, const char *s) {
  u8g2_current->drawStr(x-offset,y,s);
}

void drawWelcome() {
  u8g2_current->setFont(u8g2_font_ncenB10_tr);
  drawStr(60, 24, "Welcome");
}

void setup(void) {
    u8g2i.begin();
    u8g2.begin();
}

void loop(void) {

    setLeftSide();
    u8g2_current->firstPage();
    do { drawWelcome(); } 
    while (u8g2_current->nextPage());

    setRightSide();
    u8g2_current->firstPage();
    do { drawWelcome(); } 
    while ( u8g2_current->nextPage());
}
```

olikraus, 2024-03-15:
Nice, looks good :-)

karlo922, 2024-04-20:
Hi Oli, you can close this issue unless you want to close it when the new final version of your lib will be released containing the new constructors.

olikraus, 2024-04-27:
Great, then I will close this right now :-)

closed 2024-04-27 by olikraus

stfallet, 2025-05-14:
Hi Karlo922, Hi Olikraus,

I'm keen as well to use the front of a Mercedes AUDIO 30 APS for a project. Any chance that you can share the full sketch and hardware setup? Further did you mange the input keys as well. Any help is appreciated. Many thanks, Stefan

karlo922, 2025-05-15:
Hi stfallet, yes I've managed both keys and display itself. You can reach me on mbslk.de:
[https://www.mbslk.de/modules.php?op=modload&name=Forums&file=viewtopic&topic=102172&forum=6](url)