olikraus/u8g2 #2379: Blank bar on 256x64 SSD1322 OLED display

lucasgomesgs0, 2024-02-24:
Hi guys, I'm trying integrate a arduino nano and a 256x64 SSD1322 2.7inch OLED display using the u8g2 library, but when I upload the code to the arduino, this white bar appears on the left side and the texts is mirrored, so I set U8G2_MIRROR to the right text like the image. I tried to use SH1122 and the screen doesn’t turn on. Has anyone had this type of problem and managed to fix it?

[image]

this is my code:
`#include <U8g2lib.h>

U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI u8g2(U8G2_MIRROR, 13, 11, 10, 9, 8);

void setup(void)
{
    u8g2.begin();
}

void loop(void)
{
    u8g2.firstPage();
    do
    {
        u8g2.setFont(u8g2_font_ncenB10_tr);
        u8g2.drawStr(16, 24, "Hello World!");
    } while (u8g2.nextPage());
}`

olikraus, 2024-02-26:
hmm.. how did you solve your problem?

olikraus, 2024-02-27:
Sounds good 😊

spiderjuka, 2024-03-03:
Hi guys,
I have exactly this same problem like lucasgomesgs0 - my display is 256x64 SSD1322 2.7inch OLED after upload code to arduino white bar appears on the left side of display and the texts is mirrored.
I used fix described by ucasgomesgs0 -  in file u8x8_d_ssd1322.c I changed default_x_offset to 0x018 and second parameter of U8X8_CAA(0xa0, 0x06, 0x011), /* Set Re-Map / Dual COM Line Mode */ to 0x016.
Finally I can confirm that that fix is working!
Olikraus - but now question - is it possible to update your u8g2 library about that fix (for screen ssd1322)? Is it possible to make driver for custom ssd1322 display?

Thank you.

olikraus, 2024-03-03:
So as a (more technical) summary, this are the required changes:

Change x offset from 0x01c to 0x018:
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L260

Change this to U8X8_CAA(0xa0, 0x06, 0x016)
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L280

right?

Is there any link to the display? We should give a different name to this constructor...

olikraus, 2024-03-03:
ok... this issue is closed already, so lets continue here: https://github.com/olikraus/u8g2/issues/2386