olikraus/u8g2 #2381: Compatibility Issue: U8g2 with BigTreeTech TFT35 3.0 (ST7920 Emulation)

jeanroukas, 2024-02-26:
I am encountering an issue using the U8g2 library with a BigTreeTech TFT35 3.0 display that emulates a ST7920. While this setup functioned without problems in Marlin 2.x, U8g2 appears to have difficulties working with the emulated display.

**Details:**

Hardware:

- ESP32 (Wroom) board
- BigTreeTech TFT35 3.0 display ([Link to Hardware Pinout](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/blob/master/Hardware/BTT%20TFT35-E3%20V3.0PIN.pdf))

**Troubleshooting Steps:**

- Tested with 2 different TFT35 displays
- Switched ESP32 boards (tried also with a Arduino Leonardo)
- Changed SPI configurations (HW/SW, pins, VSPI/HSPI)
- Updated screen firmware and tested with a custom firmware, only emulation ([NoTouchFw](https://github.com/teeminus/NoTouchScreenFirmware))

**Potential Reference:**

- Klipper merge discussion about emulation differences ([Link to Discussion](https://github.com/Klipper3d/klipper/pull/3979))
- Specific commit related to emulation ([Link to Commit](https://github.com/Klipper3d/klipper/pull/3979/commits/5c91d9a16a2887fa0ebfc37ed2c712de2b19d066))

**Additional Information:**

There is a possibility that the issue is related to the state of the CS pin (high or low), which might require specific hardware control. Additionally, data preparation and synchronization might also play a role.

While I lack C++ programming knowledge, I am eager to contribute in any way possible to find a solution for this compatibility issue.

olikraus, 2024-02-26:
Difficult... many different issues might cause the system not to work as expected. Step by step isolation of those root causes is probably required.

jeanroukas, 2024-02-26:
I want to express my sincere appreciation for your responsiveness. I understand that isolating the problem can be challenging, especially given the intricate nature of the U8g2 codebase.

While I possess limited expertise in this area, I'm determined to do my part in troubleshooting. However, I'm currently unsure about the most effective approach to navigate the U8g2 code.

I would be immensely grateful if you could offer any guidance on how I can best contribute to the investigation. Perhaps you could suggest some initial steps or areas within the U8g2 files that might be relevant to explore.

Thank you once again for your time and consideration. I eagerly await any suggestions you might have, and I remain committed to aiding in resolving this compatibility issue.

olikraus, 2024-02-26:
I suggest to setup a working test environment: Maybe a ATMEGA board with EXP1 expander, then connect an original ST7920 display to it and create some test software which works.
Basically like this: https://www.youtube.com/watch?v=4CD8ERaylmY but with u8g2 instead of u8glib

After this, we know that everything works and we can plug in the BugTreeTech display and see what happens.

jeanroukas, 2024-02-26:
To further diagnose the problem, I will purchase an actual ST7920 display for direct testing. I'll keep you updated on my progress and any findings.

In the meantime, if anyone has experience using or experimenting U8g2 with Big Tree Tech displaydisplay, please share your insights in this thread.

Thanks,