olikraus/u8g2 #2386: Blank bar and text mirrored on 256x64 SSD1322 OLED display

spiderjuka, 2024-03-03:
Hi guys,
I have exactly this same problem like lucasgomesgs0 - [Blank bar on 256x64 SSD1322 OLED display](https://github.com/olikraus/u8g2/issues/2379#top)
#2379
[image]

My display is 256x64 SSD1322 2.7inch OLED after upload code to arduino white bar appears on the left side of display and the texts is mirrored.
I used fix described by ucasgomesgs0 - in file u8x8_d_ssd1322.c I changed default_x_offset to 0x018 and second parameter of U8X8_CAA(0xa0, 0x06, 0x011), /* Set Re-Map / Dual COM Line Mode */ to 0x016.
Finally I can confirm that that fix is working!
Olikraus - but now question - is it possible to update your u8g2 library about that fix (for screen ssd1322)? Is it possible to make custome driver for ssd1322 display?

Thank you.

olikraus, 2024-03-03:
So as a (more technical) summary, this are the required changes:

Change x offset from 0x01c to 0x018:
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L260

Change this to U8X8_CAA(0xa0, 0x06, 0x016)
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L280

right?

Is there any link to the display? We should give a different name to this constructor...

lucasgomesgs0, 2024-03-03:
I don’t know what the display spiderjuka bought, but I bought this one from the link:

https://a.aliexpress.com/_mLiYYnC

olikraus, 2024-03-03:
Thanks, then I think I will call the constructor "ZJY"

spiderjuka, 2024-03-03:
Hello,

It seems that I bought this same type (ZJY): 
https://www.ebay.pl/itm/404384193340.

**olikraus** - one comments: you wrote that second change need to be: "...Change this to U8X8_CAA(0xa0, 0x06, 0x016)..." - wrong.
I changed **second** parameter of init seq U8X8_CAA(0xa0, **0x06,** 0x011) to 0x016.
So new change should looks like: U8X8_CAA(0xa0, **0x016**, 0x011).

olikraus, 2024-03-03:
Ah, ok, then we have this:

Change x offset from 0x01c to 0x018:
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L260

Change this line to U8X8_CAA(0xa0, 0x016, 0x011)
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/csrc/u8x8_d_ssd1322.c#L280

right?

Is there any link to the display? We should give a different name to this constructor...

spiderjuka, 2024-03-03:
Yes - now is ok.
Thank you for your job :)

spiderjuka, 2024-03-10:
Olikraus - when your library will be updated about that issue? Does in new update will be new scrren called SSD1322 _ZJY?

Second question - can you support me to run my screen SSD1322_ZJY on ESPHome?
ESPHome has another drivers for SSD1322 - there are also two problems like in your library: display is mirrored and blank bar - wrong offset of screen. First problem I solved in file "ssd1322_base.cpp" I was  found similar line with remap fucntion:
  this->command(SSD1322_SETREMAP);
  this->data(0x14);
  this->data(0x11);
after change adress 0x14 to 0x016 screen is OK (not mirror) but still there is blank bar - wrong offset of screen.
I can't find relevant line in ESPHome driver to your seqence "default_x_offset".
How can I add * default_x_offset = */ 0x018 to ESPHome driver?
What default_x_offset does mean? Maybe is thee any other option to write that parameter in to memory of screen at ESPHome driver?
[SSD1322_esphome_drivers.zip](https://github.com/olikraus/u8g2/files/14551234/SSD1322_esphome_drivers.zip)

kuzmanzs, 2024-03-11:
Hi spiderjuka,

thanks for the suggestion. It is almost perfect. I have only thing, if you check the text after it there is a thin line.
Do you have it as well?

Olikraus big thanks for the lib.
[image]

spiderjuka, 2024-03-11:
Hello,

Try comment line 379 at file u8x8_d_ssd1322.c:
U8X8_CAA(0xb4, 0xa0, 0x005|0x0fd),	/* Display Enhancement A */ 

Please let me know if it support you?

lucasgomesgs0, 2024-03-11:
> Hi spiderjuka,
> 
> thanks for the suggestion. It is almost perfect. I have only thing, if you check the text after it there is a thin line. Do you have it as well?
> 
> Olikraus big thanks for the lib. ![ssd1322_new](https://private-user-images.githubusercontent.com/5237626/311612853-a00e5e52-e4f4-4003-a35b-20005cae5eb9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAxODY5MTAsIm5iZiI6MTcxMDE4NjYxMCwicGF0aCI6Ii81MjM3NjI2LzMxMTYxMjg1My1hMDBlNWU1Mi1lNGY0LTQwMDMtYTM1Yi0yMDAwNWNhZTVlYjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTFUMTk1MDEwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTJhZDYwOTg1OGJiMzgxMTU3M2QwNmU2ZGE1YzdhMDQ2NGViYTdhZmMzNjFmMzE1NWFmZTc5M2RhYjdkNDM0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.DS3HzFHy6s1gq2THVVAVkJiO5W77FjgVYs3GbESsQfM)

Try other font, this fix that problem for me

spiderjuka, 2024-03-11:
After comment line U8X8_CAA(0xb4, 0xa0, 0x005|0x0fd), /* Display Enhancement A */ text should be without line.
I tried it but for ESPHome driver - there are also exist "Display Enhancement A" parameter - after comment it text on screen is without line.

kuzmanzs, 2024-03-12:
yes, it helps. Thank you!

olikraus, 2024-03-14:
> Olikraus - when your library will be updated about that issue? 

Good question. I don't have time to work on u8g2 at the moment

> Does in new update will be new scrren called SSD1322 _ZJY?

I could name the new cosntructor like this...

SirusPTB, 2024-06-25:
You guys are the best thank you so much for the fixes.

olikraus, 2024-07-21:
> You guys are the best thank you so much for the fixes.

Let's see when I will have more time to work on this :-/

olikraus, 2024-09-09:
well, then... I hope that this constructor works. Closing that issue...

closed 2024-09-09 by olikraus