olikraus/u8g2 #2393: SSD1322 issue

kuzmanzs, 2024-03-06:
Hi,
I am using some more cheap SSD1322 lcd and I have strip on the left side.
I got display from here:
https://www.ebay.com/itm/134658751869
only SPI support, Support 256x64 resolution

```
U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /*SCK=*/SPI_SCK_PIN , /*MOSI=*/SPI_MOSI_PIN, /*CS=*/SPI_CS_PIN, /*DC=*/SPI_DC_PIN , SPI_RES_PIN);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font  
  u8g2.drawStr(10,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(500);
}
```

What should I configure other?
[image]

olikraus, 2024-03-08:
1) Maybe you need to activate 16 bit mode (see FAQ https://github.com/olikraus/u8g2/blob/master/doc/faq.txt)
2) try to change the hardware remap register of the display (see the SSD1322 datasheet) after `u8g2.begin()`:
`u8g2.sendF("caa", 0xa0, 0x26, 0x11);`

spiderjuka, 2024-03-10:
Hello,

Can be that you have exactly this same issue like me - see https://github.com/olikraus/u8g2/issues/2386
You need to change two things at file:u8x8_d_ssd1322.c
First try change: default_x_offset to 0x018.
Second try change second parameter of init seq U8X8_CAA(0xa0, 0x06, 0x011), /* Set Re-Map / Dual COM Line Mode */ to 0x016.
So new change should looks like: U8X8_CAA(0xa0, 0x016, 0x011).

closed 2024-03-12 by kuzmanzs