olikraus/u8g2 #2394: feature request: porting to Micropython

AlmightyOatmeal, 2024-03-07:
I was wondering if there may be much interest in porting this library to MicroPython? I have some MCU's running MP that I would love to use this library with to drive a GU128x64 VFD. I think it would be really awesome and really take MP to the next level driving displays <3

olikraus, 2024-03-08:
My knowledge on MicroPython is close to zero...

AlmightyOatmeal, 2024-03-10:
I just saw issue 829, my apologies. 

Python it might be possible to interface with objects in C so I'll see if I can give that a try ☺️

easytarget, 2024-03-11:
I'm currently working on a script+repo that takes .bdf font files and runs https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md on them. It's currently getting some finishing touches and I'll publish asap. 
It only does a limited subset of fonts, for both technical and licencing reasons, but it's still a good set with symbols etc..
I'll link here once the repo goes public.

easytarget, 2024-03-12:
Okidoki; microPython has a built-in framebuffer, but only provides a 8x8 font for use in it. Tools exist for other font choices but are a bit, err, fanatic and over complex.

There is clearly a need for a simple to use mono font collection for all the OLED and other devices out there, but none appeared in search. I guess I hadn't realised how lucky we are having U8G2 in C/Arduino land..

End result: to begin is a repo with my conversion script here:

https://github.com/easytarget/microPyEZfonts

I have **not** *yet* populated it with any  font files; I need to mmodify my script to preserve and insert comment/copyright lines from the source `.bdf ` before I am comfortable publishing the font.py files themselves.

I also want to write my own 'printStr' class for this by butchering Peter Hinches Writer class. It's waaay to complex for my needs, with seperate cursor positioning, word wrap, scroll, etc. Wonderful for some use-cases but massively overblown for simple status displays.

Check back in my repo for updates etc. 

Example run, see the README for requirements
```console
(env) owen@sam:~/microPyEZfonts$ python convert.py 
7x13.bdf: e f r n u
8x13.bdf: e f r n u
font_tiny5.bdf: e f r n u
freedoomr10r.bdf: e f r n u
helvR12.bdf: e f r n u
ncenB08.bdf: e f r n u
<snip>

Generated font files: (6)
7px : font_tiny5
    mPyEZfont_u8g2_font_tiny5_e.py
    mPyEZfont_u8g2_font_tiny5_f.py
    mPyEZfont_u8g2_font_tiny5_r.py
    mPyEZfont_u8g2_font_tiny5_n.py
    mPyEZfont_u8g2_font_tiny5_u.py
11px : ncenB08
    mPyEZfont_u8g2_ncenB08_e.py
    mPyEZfont_u8g2_ncenB08_f.py
    mPyEZfont_u8g2_ncenB08_r.py
    mPyEZfont_u8g2_ncenB08_n.py
    mPyEZfont_u8g2_ncenB08_u.py
<snip>
```

easytarget, 2024-05-08:
A quick note that I've finished my work on https://github.com/easytarget/microPyEZfonts and released/published it. 

It's a pretty simple thing, designed for simple displays and use cases. But properly documented with examples etc. 

I also note some more advanced alternatives there, plus link to general MicroPython display/utility/gui sites, it should make a good starting point for others who come here looking for a MicroPython u8g2 alternative.

olikraus, 2024-05-10:
thanks for adding the MicroPython information here :-)