olikraus/u8g2 #2400: 'SDA1' undeclared

o7-machinehum, 2024-03-13:
I'm having an issues with compilation, for some reason U8x8lib.cpp is including Wire1.

```
pio run -e target
Processing target (platform: atmelavr; board: ATmega1284P; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284P.html
PLATFORM: Atmel AVR (5.0.0) > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 128KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr-mightycore @ 3.0.0 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EthernetENC @ 2.0.4
|-- U8g2 @ 2.35.9
|-- SPI @ 1.0
|-- Wire @ 1.1
|-- ArduinoJson @ 7.0.4
|-- Watchdog @ 3.0.2
|-- EEPROM @ 2.0
|-- CRC32 @ 2.0.0
|-- ArduinoModbus @ 1.0.9
|-- se-lib-cpp
|-- ArduinoRS485 @ 1.0.5
Building in release mode
Archiving .pio/build/target/lib3a1/libEthernetENC.a
Archiving .pio/build/target/lib458/libWire.a
Compiling .pio/build/target/lib89b/Wire1/Wire1.cpp.o
Compiling .pio/build/target/lib89b/Wire1/utility/twi1.c.o
Compiling .pio/build/target/libd46/U8g2/clib/mui_u8g2.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_arc.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_bitmap.c.o
Indexing .pio/build/target/lib3a1/libEthernetENC.a
Indexing .pio/build/target/lib458/libWire.a
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_box.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_buffer.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_button.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_circle.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_cleardisplay.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_d_memory.c.o
Compiling .pio/build/target/libd46/U8g2/clib/u8g2_d_setup.c.o
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c: In function 'twi_init1':
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:89:16: error: 'SDA1' undeclared (first use in this function); did you mean 'SDA'?
   digitalWrite(SDA1, 1);
                ^~~~
                SDA
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:89:16: note: each undeclared identifier is reported only once for each function it appears in
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:90:16: error: 'SCL1' undeclared (first use in this function); did you mean 'SCL'?
   digitalWrite(SCL1, 1);
                ^~~~
                SCL
In file included from /Users/machinehum/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:99:0,
                 from /Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:26:
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:93:7: error: 'TWSR1' undeclared (first use in this function); did you mean 'TWSR'?
   cbi(TWSR1, TWPS0);
       ^
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:93:3: note: in expansion of macro 'cbi'
   cbi(TWSR1, TWPS0);
   ^~~
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:95:9: error: lvalue required as left operand of assignment
   TWBR1 = ((F_CPU / TWI_FREQ) - 16) / 2;
         ^
/Users/machinehum/.platformio/packages/framework-arduino-avr-mightycore/libraries/Wire1/src/utility/twi1.c:103:3: error: 'TWCR1' undeclared (first use in this function); did you mean 'TWSR1'?
   TWCR1 = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
[... 165 lines cut ...]
[env:target]
platform = atmelavr
board = ATmega1284P
board_build.f_cpu =  16000000L
board_build.variant = standard
monitor_speed = 9600
board_upload.speed = 115200
; platform_packages = platformio/framework-arduino-avr-mightycore@^2.2.2
build_flags = -Wall -DUIP_CONNECT_TIMEOUT=1 -DUIP_WRITE_TIMEOUT=500 -DUIPETHERNET_DEBUG_CLIENT=1

framework = arduino
build_type = release 
check_tool = cppcheck, clangtidy
test_ignore = test_app, test_service
lib_deps = 
	jandrassy/EthernetENC@^2.0.4
	olikraus/U8g2@^2.35.9
	SPI
	Wire
	ArduinoJson
	janelia-arduino/Watchdog@^3.0.0
	EEPROM
	CRC32
	ArduinoModbus
```

o7-machinehum, 2024-03-13:
FYI https://github.com/olikraus/u8g2/issues/2207

o7-machinehum, 2024-03-13:
> @olikraus The last version that works out of the box with MightyCore through PlatformIO is 2.34.5, the next version available in the [PlatformIO registry](https://registry.platformio.org/libraries/olikraus/U8g2/versions) is 2.34.18, and this version does not work, and has the same error as @o7-machinehum reported.

Yes that works thanks! I want to bisect up to where things stop working. Would you happen to know how versions are actually managed? There is only one tag in github.

MCUdude, 2024-03-13:
Now this is really weird, and maybe related to PlatformIO itself.

The followig code does not compile, hence this issue:
```c
#ifdef U8X8_HAVE_HW_I2C
#  ifdef U8X8_HAVE_HW_I2C_TEENSY3
#    include <i2c_t3.h>
#  else
#    include <Wire.h>
#    ifdef U8X8_HAVE_2ND_HW_I2C
#      if defined(MINICORE) && defined(__AVR_ATmega328PB__)
#        include <Wire1.h>
#      endif
#    endif
#  endif
#endif /* U8X8_HAVE_HW_I2C */
```

However, this does!
```c
#ifdef U8X8_HAVE_HW_I2C
#  ifdef U8X8_HAVE_HW_I2C_TEENSY3
#    include <i2c_t3.h>
#  else
#    include <Wire.h>
#    ifdef U8X8_HAVE_2ND_HW_I2C
#      if defined(MINICORE) && defined(__AVR_ATmega328PB__)
#        include <blablabla_not_present.h>
#      endif
#    endif
#  endif
#endif /* U8X8_HAVE_HW_I2C */
```

@valeros how is this even possible!

o7-machinehum, 2024-03-13:
> Now this is really weird, and maybe related to PlatformIO itself.
>
> The followig code does not compile, hence this issue:
>
> ```c
> #ifdef U8X8_HAVE_HW_I2C
> #  ifdef U8X8_HAVE_HW_I2C_TEENSY3
> #    include [...]

Hey, yeah thanks for validating me on this one. I'm seeing the exact same thing and it was driving me insane.

FYI the break is in between 2.34.5 and 2.34.18, exactly where this offending code above was introduced.

MCUdude, 2024-03-13:
@valeros Even this code snippet includes Wire1.h, and causes the compiler to fail. I have no idea how this can even happen.

```c
#if 0
#include <Wire1.h>
#endif
```

o7-machinehum, 2024-03-13:
> @valeros Even this code snippet includes Wire1.h, and causes the compiler to fail. I have no idea how this can even happen.
> 
> ```c
> #if 0
> #include <Wire1.h>
> #endif
> ```

I wonder if platformIO is doing something insane, like grepping for headers in source and compiling the corresponding .c/cpp files  based on that.

valeros, 2024-03-13:
Hi there, I've reproduced the problem, though I'm not sure how it worked before.
Anyway, a PR that fixes the issue has been filed here https://github.com/olikraus/U8g2_Arduino/pull/12. It would be great if the maintainer could take a look.

In the meantime, as a workaround, just add `lib_ldf_mode = chain+` to your `platformio.ini`.

o7-machinehum, 2024-03-14:
> lib_ldf_mode

https://docs.platformio.org/en/stable/librarymanager/ldf.html

Being completely uneducated, I think the chain+ should be the default. But I'm sure there's a good reason for it not being.

valeros, 2024-03-14:
> and why lib_ldf_mode = chain+ fixes it

For backward compatibility reasons, the default mode of the dependency finder doesn't evaluate preprocessor directives.

olikraus, 2024-03-14:
> Hi there, I've reproduced the problem, though I'm not sure how it worked before. Anyway, a PR that fixes the issue has been filed here [olikraus/U8g2_Arduino#12](https://github.com/olikraus/U8g2_Arduino/pull/12). It would be great if the maintainer could take a look.
> 
> In the meantime, as a workaround, just add `lib_ldf_mode = chain+` to your `platformio.ini`.

I have merged the PR

olikraus, 2024-03-14:
It this code causes the problem:
https://github.com/olikraus/U8g2_Arduino/blob/ca1caf364434809734e82e7d7cf66947d4e77bdf/src/U8x8lib.cpp#L54-L56

then... I don't remember where it came from. Maybe a PR from someone else.

Please let me know how to support here further.

closed 2024-03-14 by olikraus

olikraus, 2024-03-14:
Whouldn't it be better to keep the latest stable release in that json file? Currently latest stable release for u8g2 is 2.34.x

valeros, 2024-03-14:
@olikraus 

> Whouldn't it be better to keep the latest stable release in that json file? Currently latest stable release for u8g2 is 2.34.x

Isn't 2.35.10 the latest? There is a discrepancy between `library.json` and `library.properties` files, so it would be great to have these files in sync.

olikraus, 2024-03-14:
The latest official release is 2.34.22: https://github.com/olikraus/U8g2_Arduino/releases/tag/2.34.22
This release will be used by Arduino IDE. In fact Arduino IDE will use library.properties file from the release itself and not from the repository "head".

2.35.10 (currently 2.35.14) are unstable beta releases.

ivankravets, 2024-03-21:
You must keep the `version` field the same for both manifests if you want PlatformIO to pick your updates automatically.

Otherwise, you can copy the source code of the upcoming release into the temporary folder, remove `library.properties` and publish to the PlatformIO Registry manually using [pio pkg publish](https://docs.platformio.org/en/stable/core/userguide/pkg/cmd_publish.html) command.

olikraus, 2024-03-31:
hmm... I have no idea about PlatformIO and I also don't have time to figure out how to support PlatformIO. All the platformIO related code had been contributed via PR.

My goal is to support Arduino IDE. This seems to work and will be tested by this project.

I would be happy to support PlatformIO, but I can't do this.
So I see two options: Either I remove the platformIO related code or someone is able to fix this.

ivankravets, 2024-04-19:
You need to synchronize the "version" among the all manifest:
- https://github.com/olikraus/U8g2_Arduino/blob/master/library.json#L10
- https://github.com/olikraus/U8g2_Arduino/blob/master/library.properties#L2

Finally, the latest release is 2.34.22 https://github.com/olikraus/U8g2_Arduino/releases

The easiest way to fix the issue is to set "version" the same in both manifests and create a new tag/release with that version.

olikraus, 2024-08-31:
modified the release operation, so this should be fixed now (see issue https://github.com/olikraus/u8g2/issues/2402)