olikraus/u8g2 #2401: S1D15721 and ARDUINO NANO

JT-ICEMAN, 2024-03-13:
Hi.
I'm working on an ARDUINO NANO and want to connect a display with a S1D15721 controller (POLLIN Display)

The problem: After **u8g2.begin();** the system hangs.

When I changed one line in the INIT file (u8x8_d_s1d15721.c) the display works, but only with 160 columns

  /* tile_width = */ 30,		                /* width of 20*8=160 pixel (30*8 = 240) */
-->
  /* tile_width = */ 20,		                /* width of 20*8=160 pixel (30*8 = 240) */

I have no idea why the board crashed.

Can anybody help me?

Thanks 

Jürgen

JT-ICEMAN, 2024-03-14:
Hi. 
I have found the reason by myself.
The file  **u8g2_d_setup.c**  set a wrong buffer size.

In function  void u8g2_Setup_s1d15721_240x64_x(...) replace

    buf = u8g2_m_20_13_x(&tile_buf_height);

with

    buf = u8g2_m_30_20_x(&tile_buf_height);

This works

closed 2024-03-14 by JT-ICEMAN

olikraus, 2024-03-14:
Any link to the POLLIN display?

JT-ICEMAN, 2024-03-14:
https://www.pollin.de/p/varitronix-lcd-cog-vlgem1277-01-240x64-pixel-121713

JT-ICEMAN, 2024-03-14:
Nur leider grade nicht lieferbar

olikraus, 2024-03-14:
Schade... aber danke für den Link

CypaxNET, 2024-08-05:
> This works

Hi there. I can confirm this.

Changing the buffer size as Iceman described worked for me too.
I have the display and an adapter from [Shotech ](https://www.shotech.de/de/lcd-vlgem1277-01-adapter-board-5v-vcc.html) and connected it to a Wemos D1 Mini ESP module via HW SPI.
Using `U8G2_S1D15721_240X64_1_4W_HW_SPI` or `U8G2_S1D15721_240X64_2_4W_HW_SPI` only the upper 8 or 16px of the display were writeable with the other area sometimes showing random fragments.
And using the full buffer constructor `U8G2_S1D15721_240X64_F_4W_HW_SPI` caused the MCU instantly to crash and to reboot on line `u8g2.begin();`. 
The same applies for SW SPI.
To whom it may interest: the exception decoder said:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x402038c2: Print::write(char const*) at C:\Users\Phil.Phantom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Print.h line 61
EXCVADDR: 0x00000004

Decoding stack results
0x40201dd1: u8x8_d_s1d15721_240x64 at C:\Users\Phil.Phantom\Documents\Arduino\libraries\U8g2\src\clib\u8x8_d_s1d15721.c line 230
0x40203911: Print::println(char const*) at C:\Users\Phil.Phantom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 129
0x40202ce8: setup() at C:\Users\Phil.Phantom\Documents\Arduino\libraries\U8g2\src/U8g2lib.h line 266
0x40203f64: loop_wrapper() at C:\Users\Phil.Phantom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 194
```
Changing it to `buf = u8g2_m_30_20_f(&tile_buf_height);` fixed the issue:
[image]

Danke an @JT-ICEMAN für den Tipp und Danke an @olikraus für ug8lib! 👍