olikraus/u8g2 #2403: Fail to bdfconv big u8x8 font

My-point-is, 2024-03-14:
I'm trying to obtain u8x8 font from .bdf with height greater than 1 tile but converter cuts result to 1x1 tile and I get glyph array with 8 bytes per symbol. Big u8x8 fonts from "u8x8.h" are being rendered correctly but I need unicode subset.

>bdfconv -v -f 2 -m "1024-1071>128" 7x14.bdf -o u8x8_7x14_cl.c -n u8x8_7x14_cl
Parse File 7x14.bdf: 2576 glyph(s) found
Map: map_cmd_list='1024-1071>128'
Map: exclude=0 from=1024/$400 to=1071/$42f map=128/$80
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=8, h=7
CalculateMaxBBX: Encodings x=1024, y=1026, w=1024, h=1024
bf_CalculateMinMaxDWidth: dx_min=7, dx_max=7
bf_CalculateMinMaxDWidth: x_min=0, x_max=1
bf_CalculateMinMaxDWidth: Monospaced font.
bf_CalculateMaxBitFieldSize: bbx.x=1, bbx.y=2, bbx.w=4, bbx.h=4, dwidth=5
**Generate8x8Font: Tile size width=1 height=1**
Generate8x8Font: Font size 388
Monospace Statistics: Max width extention 3, average width extention 2.1
Note: For font format 2 BBX mode has been set to 3
bf_WriteU8G2CByFilename: Write file 'u8x8_7x14_cl.c'

olikraus, 2024-03-31:
> May be it would be better if bdfconv caculated minimal tw and th to enclose the entire glyph in case tw and th are not specified explicitly?

hmmm yes, but difficult.

My-point-is, 2024-04-01:
> hmmm yes, but difficult.

As alternative you could add conversion log **Note:** about **tv, th** not being specified in case it is possible to get the sign through conversion that glyphs are probably being bigger than just 1x1 in tiles.
I think it would be helpfull for other users of converter as in my case I saw log line: **Generate8x8Font: Tile size width=1 height=1** but it didn't obvious for me that I need to specify those myself.
Thank you for you great work!

olikraus, 2024-04-01:
this i could indeed do...

olikraus, 2024-09-01:
update for log output of bdfconv done...

closed 2024-09-01 by olikraus