olikraus/u8g2 #2405: Error In Displaying  Emoji On The  ST7920 128x64 LCD

DALVI12, 2024-03-17:
I Am Trying to Display Emoji On  ST7920 128X64 LCD Display with Arduino Uno. 
But when i try to do it, it just shows a blank screen with black light ON If I remove the emoji function and only keep the text function, then it works.  

PIN CONNECTION :
PIN 1 (GND) - GND
PIN 2 (VCC) -  5V
PIN 3  (V0) - POT WIPER       
PIN 4 (RS) - D10
PIN 5 (R/W) - D11
PIN 6  (E) -  D13
PIN 15 (PSB) - GND
PIN 17 (RST) -  D8
PIN 19 (BLA) - 3.3V 
PIN 20 (BLK) - GND

olikraus, 2024-03-17:
Some code would be helpful.

DALVI12, 2024-03-28:
now the case is little different it prints but not as mention in the code ;

code : 

#include <U8g2lib.h>
#include <Wire.h>
#include <Adafruit_BMP280.h>
#include <BH1750.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8); // Software SPI

#define DHT11_PIN 7
#define BMP280_ADDRESS 0x76

Adafruit_BMP280 bmp; // BMP280 sensor (I2C)
BH1750 lightMeter;    // BH1750 sensor (I2C)
DHT dht(DHT11_PIN, DHT11); // Initialize DHT sensor

const int MQ135_PIN = A0;
const int UV_PIN = A1; // Connect the UV sensor to analog pin A1
const int SOIL_MOISTURE_PIN = A2; // Connect the soil moisture sensor to analog pin A2
const int POTENTIOMETER_PIN = A3; // Connect the potentiometer to analog pin A3

unsigned long previousMillis = 0;
const long interval = 500; // interval at which to read sensors (milliseconds)

unsigned long int avgValue; // Store the average value of the sensor feedback
int buf[10], temp;

void setup() {
  Serial.begin(9600);
  while (!Serial)
    delay(100); // wait for serial monitor

  Wire.begin(); // Initialize the I2C bus

  if (!bmp.begin(BMP280_ADDRESS)) {
    Serial.println(F("Could not find a valid BMP280 sensor, check wiring or try a different address!"));
    Serial.print("SensorID was: 0x");
    Serial.println(bmp.sensorID(), 16);
    Serial.print("        ID of 0xFF probably means a bad address, a BMP 180 or BMP 085\n");
    Serial.print("   ID of 0x56-0x58 represents a BMP 280,\n");
    Serial.print("        ID of 0x60 represents a BME 280.\n");
    Serial.print("        ID of 0x61 represents a BME 680.\n");
    while (1)
      delay(10);
  }

  bmp.setSampling(Adafruit_BMP280::MODE_NORMAL, Adafruit_BMP280::SAMPLING_X2, Adafruit_BMP280::SAMPLING_X16, Adafruit_BMP280::FILTER_X16, Adafruit_BMP280::STANDBY_MS_500); // Set BMP280 settings

  lightMeter.begin(); // Initialize BH1750 sensor
  dht.begin();        // Initialize DHT sensor

  pinMode(13, OUTPUT);
  Serial.println("Ready"); // Test the serial monitor
  u8g2.begin();
}

void loop() {
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;

    // Reading temperature, pressure, and altitude from BMP280 sensor
    float tempBMP = bmp.readTemperature();
    float pressure = bmp.readPressure();
    float altitude = bmp.readAltitude(1013.25);
[... 134 lines cut ...]
}

// Function to map altitude value to emoji
char getEmojiForAltitude(float altitude) {
  if (altitude < 0) {
    return '🌊'; // Below sea level emoji
  } else if (altitude >= 0 && altitude < 100) {
    return '🏔'; // Low altitude emoji
  } else {
    return '🌌'; // High altitude emoji
  }
}

// Function to map UV index value to emoji
char getEmojiForUVIndex(float uvIndex) {
  if (uvIndex < 3) {
    return '🌑'; // Low UV index emoji
  } else if (uvIndex >= 3 && uvIndex < 6) {
    return '🌤'; // Moderate UV index emoji
  } else {
    return '☀'; // High UV index emoji
  }
}

[image]

olikraus, 2024-03-31:
Did you check whether the emoji are included in the font? Another point is, that the emoji are mapped to different glyph numbers in u8g2. 
So: Please select a font first, then carefully calculate the glyph number for the desired emoji and use the drawGlyph function to draw the emoji.

DALVI12, 2024-04-01:
where can i find the list of emoji and symbols like % with glyph number ?

olikraus, 2024-04-01:
You need to look at the fonts: https://github.com/olikraus/u8g2/wiki/fntgrp

First thing is to select a font which contains the desired glyph (emoji). Then derive the glyph number from the font picture.

Lets consider the "emoticons21" font: https://github.com/olikraus/u8g2/wiki/fntgrpu8g#emoticons21

The first emoji in the first row has the number 32 (as written at the beginning of the row).
The second emoji would then have the number 33.
The first emoji in the second row has the number 48.

You can use the "\xNN" hex notation inside a string to print the emoji, so to draw the first emoji you can use drawStr("\x20"); (0x20 is the hex code for 32)

Other emoji are part of unicode: https://github.com/olikraus/u8g2/wiki/fntgrpunifont

Maybe you also find suitable icons here: https://github.com/olikraus/u8g2/wiki/fntgrpstreamline