olikraus/u8g2 #2406: u8g2_font_tool

Dozingfiretruck, 2024-03-18:
Hi,I've written a font survival tool that supports graphics and command lines, can I add it to the intro and tools/font?

https://github.com/Dozingfiretruck/u8g2_font_tool

olikraus, 2024-03-31:
sounds cool. Is there any example to demonstrate the advantage of your tool compared to the existing u8g2 toolset?

Dozingfiretruck, 2024-04-02:
> sounds cool. Is there any example to demonstrate the advantage of your tool compared to the existing u8g2 toolset?

1，Solve the problem that the OTF2BDF tool does not handle some fonts well, resulting in the misalignment of individual characters，such as  oppo 、xiaomi font 。 zuff font , ' will display , （Very many characters）and so on .but OTF2BDF not maintenance。
2，It is very easy to ues，ttf->c 
3，It has shell version and ui version

Our company is already using the new tool, and the problem of Chinese font deviation displayed by the previous tool is still relatively serious

Dozingfiretruck, 2024-04-02:
zfull font
old tools:
[image]
new tools:
[image]
, . ' ” ，。and so on can be handled correctly

Dozingfiretruck, 2024-05-14:
Is there anything else need me to provide?

olikraus, 2024-05-15:
So your tool is a replacement for OTF2BDF and will create a .bdf file, correct?
I am not sure what exactly I shell do here. If I remember correctly, then u8g2 will use .bdf files only for Chinese fonts.

Dozingfiretruck, 2024-05-17:
> So your tool is a replacement for OTF2BDF and will create a .bdf file, correct? I am not sure what exactly I shell do here. If I remember correctly, then u8g2 will use .bdf files only for Chinese fonts.

It is the equivalent of a U8G2 dedicated font processing software, where the OTF2BDF implementation is partially rewritten to fix the above bugs introduced by OTF2BDF

olikraus, 2024-05-17:
The effort of replacing the current u8g2 font generation chain is very high and I still don't see the benefit. I have never seen any font in u8g2 where the comma is located at the wrong place. Is there any existing font in u8g2 where this is the case?

Dozingfiretruck, 2024-05-17:
Maybe you rarely test Chinese? Or Chinese doesn't use many fonts? There have been a lot of abnormalities in the TTF of non-monospaced fonts and single pixels, in fact, not only Chinese, the impact of symbols is universal, the root of the problem lies in the OTF2BDF, this problem we have also troubled for a long time to solve, so I hope to add this tool to the document to help others, not to replace the original font tool, but to have one more option to solve the existing problem, probably most non-Chinese users or users who are not sensitive to font copyright will not encounter this problem

Dozingfiretruck, 2024-05-17:
> Maybe you rarely test Chinese? Or Chinese doesn't use many fonts? There have been a lot of abnormalities in the TTF of non-monospaced fonts and single pixels, in fact, not only Chinese, the impact [...]

In addition, I can provide a more obvious ttf font that can reproduce the problem, and you should feel this problem more obviously when you test it yourself

olikraus, 2024-05-17:
Yes, such a font would be nice

Dozingfiretruck, 2024-05-18:
> Yes, such a font would be nice

[zfull-regular.zip](https://github.com/olikraus/u8g2/files/15360039/zfull-regular.zip)

olikraus, 2024-05-27:
hmm, thanks for the font. I now see the issue. I am still not sure, whether this is a font issue, but indeed it looks like a otf2bdf problem. I fixed this by adding an extra switch in cases to otf2bdf.

https://github.com/olikraus/u8g2/blob/c7497ac0ba9697b64f2927bb6d6593682a494aa1/tools/font/otf2bdf/otf2bdf.c#L884-L888

Not sure whether the fix is correct, but it will fix the problem with your font.

Dozingfiretruck, 2024-05-28:
> 1，Solve the problem that the OTF2BDF tool does not handle some fonts well, resulting in the misalignment of individual characters，such as oppo 、xiaomi font 。 zuff font , ' will display , （Very many [...]

it is bug of otf2bdf,but OTF2BDF not maintenance , so i make this tool ,In most cases, Linux users will apt and install without compiling the OTF2BDF, and may not be aware of the problem

agugu2000, 2024-05-31:
> it is bug of otf2bdf,but OTF2BDF not maintenance , so i make this tool ,In most cases, Linux users will apt and install without compiling the OTF2BDF, and may not be aware of the problem

How is your tools calculating the box parameter?I've tested it will generate a very small box that matches most of characters.But  for safety considerations, the original code enlarged the box by using the difference between the left and right boundaries. This can make the font appear exceptionally small in certain decoding environments (which is unrelated to your project).

olikraus, 2024-06-01:
> How is your tools calculating the box parameter?

I have patched otf2bdf: 
https://github.com/jirutka/otf2bdf/pull/3/commits/56a38ff6496931f433b6e211269ab79fa9ff4ca4

olikraus, 2024-06-01:
The bug fix to my local copy of otf2bdf is here: 
https://github.com/olikraus/u8g2/commit/17b0cca50de329d1897b245a0a3b077609570730

agugu2000, 2024-06-02:
> The bug fix to my local copy of otf2bdf is here: [17b0cca](https://github.com/olikraus/u8g2/commit/17b0cca50de329d1897b245a0a3b077609570730)

Sorry i didn't see this before.is it a new toos based on the original otf2bdf?i Will have a try.

olikraus, 2024-06-02:
I just added the fix view days ago based on this issue here

olikraus, 2024-08-29:
closing for now

closed 2024-08-29 by olikraus