olikraus/u8g2 #2409: ESP32 & SSD1309 - uncorrect image

Discorez, 2024-03-25:
When the image is displayed, it is shifted to the left by several pixels.
[image]

The display is connected to the ESP like this
[image]

Init line
`U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 5, /* dc=*/ 1, /* reset=*/ 3);`

How can I fix it?
Thanks.

Gopijackson, 2024-03-31:
can u give your code! me also trying !

Discorez, 2024-03-31:
Code is attached
[SSD1309_2.42_OLED.ino.zip](https://github.com/olikraus/u8g2/files/14815196/SSD1309_2.42_OLED.ino.zip)

Gopijackson, 2024-04-01:
Thanks for sharing, but im getting this issue.
how to solve this
[image]