olikraus/u8g2 #2411: SSD1309 SPI connection is causing erratic display

Gopijackson, 2024-03-28:
I've successfully flashed my code for the SSD1309 display, but it's showing irregularities in its output, as depicted in the provided image.
[image]

#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

#define OLED_SCL 13     // Define the SCL pin
#define MOSI_PIN 11     // Define the MOSI pin
#define OLED_CS  10     // Define the CS pin
#define OLED_DC   9     // Define the DC pin
#define OLED_RESET 8   // Define the RESET pin
U8G2_SSD1309_128X64_NONAME2_1_4W_SW_SPI u8g2(U8G2_R0,OLED_SCL, MOSI_PIN,/* cs=*/ OLED_CS, /* dc=*/ OLED_DC, /* reset=*/ OLED_RESET);
                        
void setup() {
  u8g2.begin();
  u8g2.setFont(u8g2_font_logisoso28_tf); // Set font size to a suitable size
  u8g2.clearBuffer();
}
void loop() {
  u8g2.drawStr(5, 28, "Hi");
  u8g2.sendBuffer();
  delay(3000);
  u8g2.clearBuffer();
  delay(1000);
}

How can I fix it?
Thanks.

olikraus, 2024-03-31:
You need to use "U8G2_SSD1309_128X64_NONAME2_F_4W_SW_SPI "

Gopijackson, 2024-04-01:
i tried that also, but still that erratic display.
plz, check that below image for that display back side that is crt right?
[image]

plz, help to solve the issuse

olikraus, 2024-04-01:
> plz, help to solve the issue

not sure how I can do this.

What is the display output if you just repace the constructor with the initially provided code?

Gopijackson, 2024-04-01:
No change,Same like first display pic

olikraus, 2024-04-01:
In your code you have used sendBuffer() to update the display. However, sendBuffer() requires the "F" constructor. In your code you have instead used the "1" constructor, which is wrong and typically it leads to a blank 8-pixel height row on the display. Such a blank 8-pixel height row seems to be there in the  lower part of your both of your pictures. Are you really sure you have tested your code with the "F" constructor?

Gopijackson, 2024-04-01:
I tested with F constructor

olikraus, 2024-04-01:
hmm maybe there is something wrong with the DC line...

Gopijackson, 2024-04-02:
plz help, how to solve this issue !

olikraus, 2024-04-14:
Not sure how I can do this from my end.