olikraus/u8g2 #2412: How to draw ellipsis with drawUTF8

JanB97, 2024-03-29:
I am using the u8g2_font_profont22_tf font.
The font has the … (ellipsis) glyph, but I am unable to use it:

[image]

I tried adding "0xE2 0x80 0xA6" to the String and I have also tried just pasting the encoded "…", but it just doesn't get displayed.

What sequence do I need to include with this font?

olikraus, 2024-03-31:
Did you enable UTF8? What is the actual code?

JanB97, 2024-04-01:
Yes, this is my setup, I have one of those OLEDs that is 2 screens in one:
[image]

The function I call is this:
`drawUTF8(22, 97, text.c_str());`

That is declared like this: 
```
void drawUTF8(u8g2_uint_t x, u8g2_uint_t y, const char *s){
  u8g2_current->drawUTF8(x-offset,y,s);
}
```
In my String i tried using: "\u2026", (0xE2,0x80,0xA6) and "…".
With "\u2026" and "…", it just is blank and with the byte codes it prints them as numbers.

These are the attempts:
[image]

text is a String and cstr is a char array with 60 elements.

olikraus, 2024-04-01:
But "..." is at position "\x85"

JanB97, 2024-04-01:
You are right, its working with "\x85", thank you so much :)
I searched everywhere but all I could find was the unicode.
I think also finally understood the font overview.

closed 2024-04-01 by JanB97