olikraus/u8g2 #2415: U8g2 Fonts wiki section, too many images causes ISP or Github to temporarily block github.com access

chriscarver, 2024-04-03:
https://github.com/olikraus/u8g2/wiki/fntlistall

This section of the wiki is only able to partially load before site access is blocked for 5 or so minutes. Unable to tell if the block is from my ISP or Github itself as of yet.

olikraus, 2024-04-03:
This is just an index page and doesn't contain much information.

chriscarver, 2024-04-04:
Sorry, linked the incorrect page, I tried just coping the link from sidebar and it gave the index instead of the page directly.

https://github.com/olikraus/u8g2/wiki/fntlist8x8

I was attempting to preview fonts to decide on a font and everytime I tried to load the page I would be blocked before all the images could load. Oddly today it is fine. I was able to reproduce this yesterday multiple times. I'd load other github pages and browse around, then swap back to the partially loaded page above and refresh, soon as I did github.com stopped working for 5 or so minutes

I'd check if github.com access was restored with some new tab at github.com, and once access was restored, I'd again randomly browse other pages to same results, once I went back to the fntlist8x8 and tried to load it, access was immediately blocked

I have no idea why or what was going on. I don't know is my ISP saw too many requests, if github saw a bunch of images and thought it was a service attack of some sort.

olikraus, 2024-04-14:
There had been issues with the wiki pages and the current split into multiple smaller pages had been the  result to avoid this problem. Basically I informed github people regarding this issue and I think at the moment it works for most users.

combs, 2025-09-28:
I'm seeing lots of temporary blocks (probably stronger filters due to all the genai crawlers these days) when I view the font list pages on the u8g2 wiki. Eventually they're all in my browser cache and I can use it without problems, but it's a bit startling to have my access to github.com temporarily blocked...