olikraus/u8g2 #2416: print Persian/Arabic letters and numeral

HoseinSL, 2024-04-04:
Hi, I use the U8G2 library on STM32, how do I write Persian/Arabic letters and numerals on the display? 
I used the below code to print the "سلام" string on display, but it printed on display like "س‌ل‌ا‌م". 
What is the problem and how should I fix it?

u8g2_FirstPage(&u8g2);
do
 {
	u8g2_SetFont(&u8g2, u8g2_font_iranian_sans_12_t_all);
	u8g2_DrawUTF8(&u8g2, 40, 10, "سلام");
  } while (u8g2_NextPage(&u8g2));

olikraus, 2024-04-14:
This is a difficult issue. It had been discussed multiple times, but as of now there is no proper solution in u8g2 itself. I suggest to look into the existing issues, maybe there is something useful.

https://github.com/olikraus/u8g2/issues?q=is%3Aissue+arabic

Especially: https://github.com/olikraus/u8g2/issues/640, https://github.com/olikraus/u8g2/issues/529, https://github.com/olikraus/u8g2/issues/1015

olikraus, 2024-05-10:
Nice, thanks for the feedback (link repeated: https://github.com/idreamsi/u8g2-persian-reshaper)