olikraus/u8g2 #2418: Any news from grayscale?

Adamyno, 2024-04-08:
I have an SSD1325 OLED display, which according to the data sheet can handle (16) levels of grayscale. This is not entirely clear to me because the original u8glib could control 4 different shades of gray on 2 bits. Unfortunately, I couldn't find a usable u8glib for an ESP32 board. The newer u8g2 can now be used with the ESP32 board, but it cannot handle grayscale. Thus, the display is essentially a waste, because it can do the same in this form as an SSD1306.

olikraus, 2024-04-14:
Looks like also the Adafruit lib just supports monochrome graphics: https://reference.arduino.cc/reference/en/libraries/adafruit-ssd1325/

But to answer your question: No, nothing new from my end.

Hrishikesh46779, 2025-02-08:
Will we see grayscale library anytime soon?

olikraus, 2025-02-09:
Too less time for this from my end. :-/