olikraus/u8g2 #2420: RaspPi Pico/Arduino IDE: Can't get the library to use other Clock and Data Pins.

theghostglitch, 2024-04-09:
My SSD1306 seems very finicky. I can't get it to use any other driver then U8G2_SSD1306_128X64_NONAME_F_HW_I2C
This driver seems to ignore any input for Clock and Data pins.

I've used this constructor U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 21, 20);
but it defaults to pins 5 and 4. According to the Pico documentation, these sets of pins are on the same I2C controller as the defaults so this should work. Am I missing something?

olikraus, 2024-04-14:
Use "U8G2_SSD1306_128X64_NONAME_F_SW_I2C"

egnor, 2024-04-14:
See https://github.com/olikraus/u8g2/issues/2289 (same issue?). (Software I2C is slowwwww!)

olikraus, 2024-04-14:
The pin renaming feature as used in the initial post here, is only available for ESP systems. U8g2 simply does not know the SCL / SDA pins and assigning any other values is a non-standard Arduino extension.
But I think the suggestion from @egnor should work.

theghostglitch, 2024-04-14:
For some reason "U8G2_SSD1306_128X64_NONAME_F_SW_I2C" doesn't work at all on this display? No clue why. But yes, @egnor 's solution works. Thanks.

closed 2024-04-14 by theghostglitch