olikraus/u8g2 #2421: Problem with PCF8812

bnv12345, 2024-04-09:
Hello. I'm trying to start a display from Siemens C55 (PCF8812).  The display only worked when I replaced it in the file u8x8_d_pcf8812.c Initialization sequence.

_static const uint8_t u8x8_d_pcf8812_96x65_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x21), // PowerON, ExtCommandSet
  U8X8_C(0x09), // Internal HV-gen x3
  U8X8_C(0xB7), // Set Vop
  U8X8_C(0x16), // Bias n=2    //15
  U8X8_C(0x06), // Temperature coeff 2
  U8X8_C(0x20), // StandartCommandSet
  U8X8_C(0x0C), // normal mode, display non-inverted
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};_

I found this information on the Internet. Long-standing problem. The resolution is not yet specified correctly. It was also in the first version of this library.
Also changed the screen resolution-

 _static const u8x8_display_info_t u8x8_pcf8812_96x65_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 100,
  /* pre_chip_disable_wait_ns = */ 100,
  /* reset_pulse_width_ms = */ 100, 
  /* post_reset_wait_ms = */ 100, 
  /* sda_setup_time_ns = */ 100,	
  /* sck_pulse_width_ns = */ 100,	
  /* sck_clock_hz = */ 4000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 150,	
  /* tile_width = */ 12,
  /* tile_height = */ 9,
  /* default_x_offset = */ 0,                    
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 101,                        // 96
  /* pixel_height = */ 64
};_

Everything worked, but the five right columns are garbage.

[image]

If I reduce this parameter (/* tile_width = */ 12) to 11, then there is more garbage.
[image]

 If you put 13 then there is no image. Please, help.

olikraus, 2024-04-14:
Increasing the tile size will cause memory issues and indeed might lead to a reboot.

What is the actual pixel width of your display?

bnv12345, 2024-04-14:
This is a display from Siemens C55.  Actual pixel size is  101*64

olikraus, 2024-04-14:
I have created a new constructor "U8G2_PCF8812_101X64".
Can you do a test, whether this works better?
The code is available in the latest beta release 2.35.17

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

bnv12345, 2024-04-16:
The new library did not work because the display initialization was not changed. In the file u8x8_d_pcf8812.c you need to put

static const uint8_t u8x8_d_pcf8812_96x65_init_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x21), // PowerON, ExtCommandSet
  U8X8_C(0x09), // Internal HV-gen x3
  U8X8_C(0xB7), // Set Vop
  U8X8_C(0x16), // Bias n=2    //15
  U8X8_C(0x06), // Temperature coeff 2
  U8X8_C(0x20), // StandartCommandSet
  U8X8_C(0x0C), // normal mode, display non-inverted
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

I wrote this in my first message. After the replacement this code everything worked great. There is no garbage.

olikraus, 2024-04-27:
Thanks for testing. I have added the missing init code in beta release 2.35.18

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

bnv12345, 2024-04-28:
Unfortunately, this option without editing did not work either. I edited the `u8x8_d_pcf8812.c` file a little and everything worked again.
 Added a function -` static uint8_t u8x8_d_pcf8812_101x64_generic(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)`
And the last line of this file
 `return u8x8_d_pcf8812_96x65_generic(u8x8, msg, arg_int, arg_ptr); ` replaced with 
`return u8x8_d_pcf8812_101x64_generic(u8x8, msg, arg_int, arg_ptr);`

bnv12345, 2024-04-28:
```/*

  u8x8_d_pcf8812.c
  
  pcf8812: 65x102
  pcf8814: 65x96

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2017, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  
  
*/

#include "u8x8.h"

static const uint8_t u8x8_d_pcf8812_96x65_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x020),		                /* activate chip (PD=0), horizontal increment (V=0), enter normal command set (H=0) */
  U8X8_C(0x008),		                /* blank display */
  
  U8X8_C(0x021),		                /* activate chip (PD=0), horizontal increment (V=0), enter extended command set (H=1) */
  U8X8_C(0x006),		                /* temp. control: b10 = 2 */
  U8X8_C(0x013),		                /* bias system, 0x010..0x07 1:48 */
  U8X8_C(0x09f),				/* contrast setting, 0..127 */
  //U8X8_CA(0x020 | 2, 0x080 | 0),				/* contrast setting, pcf8814 */

  U8X8_C(0x024),		                /* deactivate chip (PD=1), horizontal increment (V=0), enter normal command set (H=0) */
    
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_pcf8812_101x64_init_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x21), // PowerON, ExtCommandSet
  U8X8_C(0x09), // Internal HV-gen x3
  U8X8_C(0xB7), // Set Vop
  U8X8_C(0x16), // Bias n=2    //15
  U8X8_C(0x06), // Temperature coeff 2
  U8X8_C(0x20), // StandartCommandSet
  U8X8_C(0x0C), // normal mode, display non-inverted
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
[... 223 lines cut ...]
  /* write_pulse_width_ns = */ 150,	
  /* tile_width = */ 13,
  /* tile_height = */ 8,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 101,
  /* pixel_height = */ 64
};

uint8_t u8x8_d_pcf8812_101x64(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    if ( msg == U8X8_MSG_DISPLAY_SETUP_MEMORY )
    {
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_pcf8812_101x64_display_info);
      return 1;
    }
    else if ( msg == U8X8_MSG_DISPLAY_INIT )
    {
      u8x8_d_helper_display_init(u8x8);
      u8x8_cad_SendSequence(u8x8, u8x8_d_pcf8812_101x64_init_seq);    
    }    
    return u8x8_d_pcf8812_101x64_generic(u8x8, msg, arg_int, arg_ptr);
}

```

olikraus, 2024-04-28:
create beta 2.35.19...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

bnv12345, 2024-04-30:
Yes. Now everything works as it should. Thank you.

closed 2024-05-20 by bnv12345